AdnanHodzic/auto-cpufreq #297: [Suggestion] Add configuration options based on temperature

MithicSpirit, 2021-12-02:
It would be helpful if there was a way to reduce to `powersave` when the CPU is above a certain temperature to prevent overheating.

AdnanHodzic, 2021-12-08:
Might be an interesting feature to have indeed, but could you please try out latest changes? With [release of 1.8.0](https://github.com/AdnanHodzic/auto-cpufreq/releases) I think overheating should be drastically lowered to non existent now.

MithicSpirit, 2021-12-09:
I am still having issues of very high temps when performance governor + boost clock while charging (though this may be related to me using tlp to set the CPU max frequency, I'll do some more testing on this later).

AdnanHodzic, 2021-12-09:
As specified in [project readme](https://github.com/AdnanHodzic/auto-cpufreq) "Please note: auto-cpufreq aims to replace TLP and after you install auto-cpufreq it's recommended to remove TLP. However, if for some reason you still need/want TLP installed and running auto-cpufreq doesn't conflict and works great in tandem with TLP."

> though this may be related to me using tlp to set the CPU max frequency, I'll do some more testing on this later)

I'm sure TLP is the reason, you have a conflict between auto-cpufreq and TLP on who's going to set what.

MithicSpirit, 2021-12-09:
Disabling tlp has fixed the issue, though now I can't control the maximum allowed frequency when not charging, which somewhat decreases my battery life. So I think that there should be another option in the config to allow specifying the maximum (and maybe minimum) frequencies. 
I can open another issue for this if you want.

AdnanHodzic, 2021-12-13:
> Disabling tlp has fixed the issue,

[With release of 1.8.1](https://github.com/AdnanHodzic/auto-cpufreq/releases) user will get a warning message if auto-cpufreq detects TLP is running in background.

> though now I can't control the maximum allowed frequency when not charging, which somewhat decreases my battery life. So I think that there should be another option in the config to allow specifying the maximum (and maybe minimum) frequencies.
> I can open another issue for this if you want.

Yes, please do, in meantime I'll close this one.

closed 2021-12-13 by AdnanHodzic