AdnanHodzic/auto-cpufreq #307: Doesn’t change to perform when plugged in

ShabirK21, 2021-12-13:
So I have a Lenovo laptop which has conservative mode, when I plug in with it on, auto-cpufreq doesn't switch to performance and stays in powersave scaling

bobslept, 2021-12-13:
This issue is probably the same as #299
I'm not really sure what is going on, because I don't experience this issue myself. I'm on a lenovo which uses the ideapad_laptop kernel module. But it stays in charging state when I am on 60% when using conservative mode.

ShabirK21, 2021-12-13:
Then it should be due my system battery status being read as unknown. Thanks anyway

bobslept, 2021-12-13:
Are you comfortable with git and want to test something out?

ShabirK21, 2021-12-14:
So i tested out the program, and i'm glad to say it works! when my bat status is unknown, the live status reads it as charging and sets the "performance" & switches to "powersave" when unplugged. Awesome! thanks for the fix

bobslept, 2021-12-14:
Thank you so much for testing this. Very happy to hear it solved your problem! 
We probably need a little bit more users to test this before we actually merge into master.
Thanks again!

AdnanHodzic, 2021-12-16:
Thanks to @bobslept the proposed changes have now been merged with `master` branch. Changes will be part of the upcoming 1.8.2 release.  

Until the release is made, changes can be installed by either having latest `master` changes and installing them using [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto-cpufreq-installer) or using `beta` channel of Snap package (`sudo snap install auto-cpufreq --beta`). With that said I'm closing this issue, please re-open it if the issue persists.

closed 2021-12-16 by AdnanHodzic