AdnanHodzic/auto-cpufreq #309: Set max/min allowed frequency

MithicSpirit, 2021-12-13:
It would be nice to add options to set the min/max frequencies on AC or battery since these are available on tlp.
(Continued from #297)

AdnanHodzic, 2021-12-16:
It would be great to have this functionality and it would be ideal for the [config file](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq). 

Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution).

varaki, 2021-12-24:
#324

AdnanHodzic, 2021-12-26:
This issue has now been resolved as part of [v1.9.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

closed 2021-12-26 by AdnanHodzic