AdnanHodzic/auto-cpufreq #313: Advised method for disabling gnome power management gives error

Woodhousemelena, 2021-12-17:
### Error output:
```text
Traceback (most recent call last):
  File "/home/xxxx/auto-cpufreq/auto_cpufreq/power_helper.py", line 9, in <module>
    from auto_cpufreq.tlp_stat_parser import TLPStatusParser
ModuleNotFoundError: No module named 'auto_cpufreq.tlp_stat_parser'

```
---

### System information:
```text
Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: Manjaro Linux 21.2.0 Qonos
Linux kernel: 5.14.18-1-MANJARO
Processor: AMD Ryzen 7 4700U with Radeon Graphics
Cores: 8
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2000 MHz
CPU min frequency: 1400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  4.0%     62 °C     2000 MHz
CPU1:	  5.1%     62 °C     2000 MHz
CPU2:	  3.0%     62 °C     2000 MHz
CPU3:	  5.0%     62 °C     2296 MHz
CPU4:	  4.0%     62 °C     2000 MHz
CPU5:	  3.0%     62 °C     2000 MHz
CPU6:	  2.0%     62 °C     2000 MHz
CPU7:	  1.0%     62 °C     2000 MHz

auto-cpufreq version:
Version         : 1.8.0-1

Python: 3.9.7
psutil package: 5.8.0
platform package: 1.0.8
click package: 8.0.3
distro package 1.6.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.12 %

Total CPU usage: 7.0 %
Total system load: 0.88
Average temp. of all cores: 61.625 °C 

Currently using: performance governor
Currently turbo boost is: on

-------------------------------------------------------------------------------

```
---

bobslept, 2021-12-17:
Which command did you run? And how did you install auto-cpufreq?
Because I can't reproduce this error myself.

AdnanHodzic, 2021-12-19:
@Woodhousemelena can you confirmed you're experiencing this problem on [1.8.2](https://github.com/AdnanHodzic/auto-cpufreq/releases)?

AdnanHodzic, 2022-02-20:
Just [released v1.9.2](https://github.com/AdnanHodzic/auto-cpufreq/releases) which adds ability to change power governor used by default with auto-cpufreq from `balanced` to `performance` [using power_helper.py script](https://github.com/AdnanHodzic/auto-cpufreq/#1-power_helperpy-script) which could also resolve this problem.

In meantime, closing the issue due to inactivity, if the problem persist please feel free to re-open the issue.

closed 2022-02-20 by AdnanHodzic