AdnanHodzic/auto-cpufreq #315: [Request] Change CPU governor based on battery level

mizzunet, 2021-12-19:
Currently `auto-cpufreq` changes governor according to power state (Charging and Discharging), right?

I think it would be awesome to be able to set governor based on battery levels as well. 
For example,
Charging state governor: `performance`
Discharging state governor: `schedutil`
If battery is below 40%, governor: `powersave`

Thanks

josh1244, 2022-01-14:
I agree, this would be a good feature. Is there a way to manually control the govener?

mizzunet, 2022-01-14:
> Is there a way to manually control the govener?

Yeah, by editing `/etc/auto-cpufreq.conf`

AdnanHodzic, 2022-11-11:
Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).