AdnanHodzic/auto-cpufreq #323: PopOS pull request Snap permission denied

tyleraharrison, 2021-12-20:
I just made the pull request that fixed #276 but after updating through the Snap repository, I get this:

```
Warning: Cannot get distro name
Pop!_OS detected
Pop!_OS uses a symbolic link for the os-release file, this causes issues and can be fixed by converting to a hard link
Attempting to change symlink to hard link for /etc/os-release -> /etc/pop-os/os-release
Continue? [y/N] y
Creating hard link for /etc/os-release
sh: 1: sudo: Permission denied
sh: 1: sudo: Permission denied
```
After some searching, it seems that Snap has issues running sudo from `os.system`. You may want to remove PR #311 and release 1.8.2 until this issue is resolved

tyleraharrison, 2021-12-20:
I'll be at work all day, it's weird that it worked for me in my testing environment but not after pulling from the Snap repository this morning. Sorry for the faulty PR, if I get off of work today and this is not resolved then I will fix it myself

bobslept, 2021-12-20:
I see, snap container does it's job when touching the files I think. Thanks for reporting this.

@AdnanHodzic maybe you know if this can be fixed, or that we have to exclude this option when running from inside snap.

AdnanHodzic, 2021-12-21:
During #276 review process I somehow overlooked that this would be a problem when auto-cpufreq is installed as a Snap package. As in this case, auto-cpufreq is running as part of a container and doesn't have access to host system files. Hence, I just sent a [request for system-files interfaces (hardlink to /etc/os-release)](https://forum.snapcraft.io/t/request-for-system-files-interfaces-hardlink-to-etc-os-release-for-auto-cpufreq/28015) with my proposal how this problem should be fixed. 

After this request is approved I would push proposed changes to `snap/snapcraft.yaml`, updating existing or create a new release and problem would be fixed. In meantime, let's see what Canonical people say.

bobslept, 2021-12-21:
Is this only an issue on PopOS 20.04?
I'm just installed PopOS 21.10 in a virtual machine and I can just do a distro.id() in a python shell without any problems.

```
>>> import distro
>>> distro.id()
'pop'
>>> distro.name()
'Pop!_OS'
>>> 
beep@pop-os:~/auto-cpufreq$ cat /etc/os-release
NAME="Pop!_OS"
VERSION="21.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 21.10"
VERSION_ID="21.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=impish
UBUNTU_CODENAME=impish
LOGO=distributor-logo-pop-os
beep@pop-os:~/auto-cpufreq$ 
 ```

bobslept, 2021-12-21:
Without changes on a fresh PopOS 20.04 VM I could just `import distro`, and do and `distro.id()` without any problems.

So is it just that the Snap package that would not follow a softlink to /etc/pop-os/os-release because of the container?

bobslept, 2021-12-21:
I think that without snap it should just work on PopOS. Even without creating new hardlink.
I've installed 20.04 and 21.10 and when opening a python shell I can just use `distro` package and it just returns good values.
So maybe adding just read access to `/etc/pop-os/os-release` and `/etc/os-release` would work?
Just thinking out loud.

bobslept, 2021-12-21:
@tyleraharrison I have one question. If you make this hardlink yourself, outside the snap. And then run the snap. Do you get the same error? 

Are we 100% sure that hardlink `/etc/pop-os/os-release` solve the issue when using a snap? Before we do all kind of stuff to make an hardlink that in the end doesn't even fix the permission denied error.

tyleraharrison, 2021-12-21:
@bobslept Yes, I confirmed that even with the Snap install, doing the hard link manually fixes it for me. PopOS is my daily driver

bobslept, 2021-12-21:
> @bobslept Yes, I confirmed that even with the Snap install, doing the hard link manually fixes it for me. PopOS is my daily driver

Ok, thank you :+1:

AdnanHodzic, 2021-12-22:
> If we really have issues creating this stuff from snap context, we could always print a message that they have to create the hardlink themselfs with instructions and then exit. Instead of the automatic route. We will see what happends.

I was about to say this in my original message. In case [Canonical denies my request](https://forum.snapcraft.io/t/request-for-system-files-interfaces-hardlink-to-etc-os-release-for-auto-cpufreq/28015) since `/etc/os-release` is not a file own by auto-cpufreq. A simple change where if Snap and Pop OS is detected outputs how to do this manually.

tyleraharrison, 2021-12-22:
I am writing a solution right now

tyleraharrison, 2021-12-22:
@AdnanHodzic How does this look? Also what is your preferred way to detect if this is a Snap or not?

If the program is not a Snap:
```
Warning: Cannot get distro name
Pop!_OS detected
Pop!_OS uses a symbolic link for the os-release file, this causes issues and can be fixed by converting to a hard link
You can change the symbolic link to a hard link for /etc/os-release -> /etc/pop-os/os-release
Continue? [y/N] y

Backing up /etc/os-release to /etc/os-release-backup
Creating hard link for /etc/os-release

Checking to see if solution worked...
Success!
```

If the program is a Snap:
```
Warning: Cannot get distro name
Pop!_OS detected
Pop!_OS uses a symbolic link for the os-release file, this causes issues and can be fixed by converting to a hard link
You can change the symbolic link to a hard link for /etc/os-release -> /etc/pop-os/os-release
Continue? [y/N] y

Snap install detected, you must manually run the following commands in another terminal:

Backup the /etc/os-release file:
sudo mv /etc/os-release /etc/os-release-backup

Press [Enter] to continue...

Create hard link to /etc/os-release:
sudo ln /etc/pop-os/os-release /etc/os-release

Press [Enter] to continue...

Checking to see if solution worked...
Success!
```

In the event the solution did not work:
```
Backing up /etc/os-release to /etc/os-release-backup
Creating hard link for /etc/os-release

Checking to see if solution worked...
Failed, aborting...
```

If the OS cannot be read and the operating system is NOT PopOS:
```
Warning: Cannot get distro name
Check /etc/os-release permissions and make sure it is not a symbolic link
Aborting...
```

tyleraharrison, 2021-12-22:
The only issue I see this causing is automated programs that may try to install this and get hung up at those user inputs. I can remove them, but I really liked that doing it this way allows for the program to check once again and continue if everything was done correctly by the user instead of having the user restart the program

tyleraharrison, 2021-12-22:
Feel free to add the Snap detection and any other changes to this pull request, it does not have the detection in it's current state, just a dummy variable

tyleraharrison, 2021-12-22:
Also, with those plugs just now committed to 1.9.0, will this still build for others if they pull it right now since Canonical hasn't implemented this yet?

bobslept, 2021-12-22:
Isn't it a snap only issue? After reading this comment from #240
> I got this error too when installing from Snap on Pop!OS 21.04. I was able to work around by installing with [git instead](https://github.com/AdnanHodzic/auto-cpufreq#auto-cpufreq-installer).
I've installed a VM with PopOS and the `distro` package from a python shell works just fine without changes.

If so it would be lot easier. On permission denied, detect file exists, detect snap, print solution.

I think snap detection is like this:
```
if os.getenv("PKG_MARKER") == "SNAP":
   # we are in snap package
```

AdnanHodzic, 2021-12-22:
@tyleraharrison proposed solution how the output would look like good, but @bobslept also summarized it perfectly in his last message. 

> Also, with those plugs just now committed to 1.9.0, will this still build for others if they pull it right now since Canonical hasn't implemented this yet?

No, since those plugs were not approved by Canonical, 1.9.0 release upload to beta channel didn't go through and it triggered a manual review by one of the staff members. As part of feedback I pointed out that request has been made for this and [linked](https://forum.snapcraft.io/t/request-for-system-files-interfaces-hardlink-to-etc-os-release-for-auto-cpufreq/28015) to that request for whoever reviews it from their side can see it.

bobslept, 2021-12-22:
Let's go and just print a message then.

tyleraharrison, 2021-12-24:
If it's OK I believe this issue can be closed and potentially revisited if Canonical responds

AdnanHodzic, 2021-12-26:
> If it's OK I believe this issue can be closed and potentially revisited if Canonical responds

They will respond, but I'm pretty sure they will deny the request, as `/etc/os-release` is not a solely auto-cpufreq owned file. 

Regardless, until we hear it from them I'm closing the issue as changes live as part of [v1.9.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

AdnanHodzic, 2022-01-03:
> If it's OK I believe this issue can be closed and potentially revisited if Canonical responds

@tyleraharrison response from Canonical is [here](https://forum.snapcraft.io/t/request-for-system-files-interfaces-hardlink-to-etc-os-release-for-auto-cpufreq/28015), seems like there is a way to get this working properly with Pop OS.

bobslept, 2022-01-03:
How does this response from Canonical change the fact that we cannot follow the soft link from the "readable" /etc/os-release?

First result is with hardlink, second is with default softlink
```
beep@pop-os:~/auto-cpufreq$ snap run --shell auto-cpufreq
To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.
 
beep@pop-os:/home/beep/auto-cpufreq$ cat /var/lib/snapd/hostfs/etc/os-release
NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os
beep@pop-os:/home/beep/auto-cpufreq$ cat /var/lib/snapd/hostfs/etc/os-release
cat: /var/lib/snapd/hostfs/etc/os-release: Permission denied
beep@pop-os:/home/beep/auto-cpufreq$ 
 ```
Am I missing something?

AdnanHodzic, 2022-01-03:
@bobslept that's a valid question, I asked them to clarify.

tyleraharrison, 2022-01-03:
Awaiting clarification

bobslept, 2022-01-03:
Well I have left a message on the PopOS issue tracker about our issue.

AdnanHodzic, 2022-01-03:
Thanks @bobslept, based on your answer I assume you saw the clarification :slightly_smiling_face: @tyleraharrison see the last [message](https://forum.snapcraft.io/t/request-for-system-files-interfaces-hardlink-to-etc-os-release-for-auto-cpufreq/28015/3).

tyleraharrison, 2022-01-03:
That was a very verbose "no you can't" but I like that. Is there a way to just read the PopOS release file instead of modifying the link?

bobslept, 2022-01-03:
Well that doesn't work on a SNAP package. That gives us the information of the snap container, and not host info. At least it gives different results on my side.

On the host
```
beep@pop-os:~$ cat /usr/lib/os-release
NAME="Pop!_OS"
VERSION="20.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
LOGO=distributor-logo-pop-os
```
Inside a snap shell
```
beep@pop-os:/home/beep$ cat /usr/lib/os-release
NAME="Ubuntu Core"
VERSION="20"
ID=ubuntu-core
PRETTY_NAME="Ubuntu Core 20"
VERSION_ID="20"
HOME_URL="https://snapcraft.io/"
BUG_REPORT_URL="https://bugs.launchpad.net/snappy/"
```

tyleraharrison, 2022-01-04:
That was a typo, I meant /etc/pop-os/os-release

AdnanHodzic, 2022-01-05:
Let's just leave it as is for now, it's waste of energy and time especially since we implemented a workaround. 

I also left this [as a reply,](https://forum.snapcraft.io/t/request-for-system-files-interfaces-hardlink-to-etc-os-release-for-auto-cpufreq/28015/7?u=ahodzic) and they have already started not approving it as I previously predicted:

> They will respond, but I'm pretty sure they will deny the request, as `/etc/os-release` is not a solely auto-cpufreq owned file.

closed 2022-01-05 by AdnanHodzic