AdnanHodzic/auto-cpufreq #331: Battery reported as discharging even it's not

mhalano, 2021-12-24:
I'm having problems with detection of charging state. My laptop is connected to AC and fully charged (as reported by GNOME) but still get `discharging` and `powersave` is enabled. It happend to 1.9.0 and I downgraded to 1.8.2 and still happening.

### System information:
```text
-------------------------------------------------------------------------------

Linux distro: Ubuntu 21.10 (Impish Indri)
Linux kernel: 5.15.10-051510-generic
Processor: 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz
Cores: 16
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2300 MHz
CPU min frequency: 800 MHz

Core	Usage	Temperature	Frequency
CPU0:	  2.0%     46 °C     2300 MHz
CPU1:	  2.0%     49 °C     2300 MHz
CPU2:	  3.0%     45 °C     2300 MHz
CPU3:	  5.1%     46 °C     2300 MHz
CPU4:	  2.0%     47 °C     2300 MHz
CPU5:	  2.0%     45 °C     1432 MHz
CPU6:	 12.0%     48 °C     2300 MHz
CPU7:	  3.0%     45 °C     2300 MHz
CPU8:	  1.0%     46 °C     1268 MHz
CPU9:	  2.0%     49 °C     2300 MHz
CPU10:	  3.0%     45 °C     2300 MHz
CPU11:	  1.0%     46 °C     1600 MHz
CPU12:	  4.0%     47 °C     2300 MHz
CPU13:	  6.0%     45 °C     1951 MHz
CPU14:	  2.0%     48 °C     2300 MHz
CPU15:	  5.0%     45 °C     2300 MHz

auto-cpufreq version:
Snap: 1.8.2

Python: 3.8.10
psutil package: 5.8.0
platform package: 1.0.8
click package: 8.0.3
Traceback (most recent call last):
  File "/snap/auto-cpufreq/96/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/snap/auto-cpufreq/96/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/snap/auto-cpufreq/96/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 9, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/snap/auto-cpufreq/96/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 7, in <module>
    from pip._internal.cli import cmdoptions
  File "/snap/auto-cpufreq/96/lib/python3.8/site-packages/pip/_internal/cli/cmdoptions.py", line 24, in <module>
    from pip._internal.exceptions import CommandError
  File "/snap/auto-cpufreq/96/lib/python3.8/site-packages/pip/_internal/exceptions.py", line 10, in <module>
    from pip._vendor.six import iteritems
ModuleNotFoundError: No module named 'pip._vendor.six'
distro package

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.05 %

Total CPU usage: 2.1 %
Total system load: 0.93
Average temp. of all cores: 46.375 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```

bobslept, 2021-12-24:
Thank you for your bug report.
Could give me the output of the following command:

```
ls /sys/class/power_supply/*
```

AdnanHodzic, 2021-12-24:
@bobslept ^

bobslept, 2021-12-24:
I'm already on it :smile:

AdnanHodzic, 2021-12-24:
> I'm already on it 😄

I think we typed that at same moment 😁

bobslept, 2021-12-24:
@mhalano I hope this issue is the same as we already fixed in master. So I'm really interested in the output of that command.

mhalano, 2021-12-24:
> Thank you for your bug report. Could you please give me the output of the following command:
> 
> ```
> ls /sys/class/power_supply/*
> ```
[image]

bobslept, 2021-12-24:
Alright thank you very much.
Could you now post the output of:
```
cat /sys/class/power_supply/ucsi*/type
cat /sys/class/power_supply/AC*/type
cat /sys/class/power_supply/hid*/type
```

bobslept, 2021-12-24:
I think I already have a solution.

Could you please try this fix.
Be sure that current install of auto-cpufreq is removed.

```
git clone https://github.com/bobslept/auto-cpufreq.git
cd auto-cpufreq
git checkout power-supply-ignore-list
sudo ./auto-cpufreq-installer --install
sudo auto-cpufreq --live
```

To remove this from your system:
```
sudo ./auto-cpufreq-installer --remove
```

mhalano, 2021-12-24:
> cat /sys/class/power_supply/ucsi*/type
> cat /sys/class/power_supply/AC*/type
> cat /sys/class/power_supply/hid*/type
> cat /sys/class/power_supply/BAT*/type

[image]

bobslept, 2021-12-24:
Thank you very much. Could you please test the fix I've created?

mhalano, 2021-12-24:
> git clone https://github.com/bobslept/auto-cpufreq.git
> cd auto-cpufreq
> git checkout power-supply-ignore-list
> sudo ./auto-cpufreq-installer --install
> sudo auto-cpufreq --live

Worked like a charm. I will wait the next stable version to release this fix, test it again and then close the issue. Seems ok to you?

bobslept, 2021-12-24:
Yes that is oke with me. I will do pull request, so it can be in next release.
Thank you very much for testing!

mhalano, 2021-12-24:
> Yes that is oke with me. I will do pull request, so it can be in next release. Thank you very much for testing!

Thank you very much for fixing. :) Have a nice holiday!

bobslept, 2021-12-24:
> > Yes that is oke with me. I will do pull request, so it can be in next release. Thank you very much for testing!
> 
> Thank you very much for fixing. :) Have a nice holiday!
No problemo, you too have a nice holiday! 
I now see that it maybe auto closes after merging into master. But if you experience the same issue on the new release you can re-open this issue.

closed 2021-12-24 by AdnanHodzic

AdnanHodzic, 2021-12-26:
Changes live as part of [v1.9.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

PR #332 (2021-12-24): Improve charging detection function
The list of items in power_supply is now sorted. This way the AC is 'always' on top, and checked first. What I noticed is that internal supplies are always use uppercase, and inserted devices all use lowercase. 

So if this theory is correct the AC and internal battery will always be checked first. 

I also noticed the "scope" file under the hid device with battery. Maybe that file could help identify if an battery is from some other device or not. Will see in future bug reports if that is correct and maybe use that.

If this still not works correctly I've i [... cut]