AdnanHodzic/auto-cpufreq #335: Pop OS doesn't have Grub to disable intel_pstate

DanielAbrante, 2021-12-26:
I read the *Troubleshooting* section and my Pop OS does not have **Grub** to disable.

Previous, i used Manjaro and everything worked fine, but i needed switch to Pop OS.

How i disable since its not working this solution?

---
### Error output: 
``` 
cat: /etc/default/grub: No such file or directory
```
---
### System information:
```text
-------------------------------------------------------------------------------

Linux distro: Pop!_OS 21.10 impish
Linux kernel: 5.15.8-76051508-generic
Processor: Intel(R) Core(TM) i5-8300H CPU @ 2.30GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2300 MHz
CPU min frequency: 800 MHz

Core	Usage	Temperature	Frequency
CPU0:	  1.0%     42 °C     2300 MHz
CPU1:	  2.1%     46 °C     2300 MHz
CPU2:	  4.9%     40 °C      900 MHz
CPU3:	  0.0%     42 °C     2300 MHz
CPU4:	  5.1%     42 °C     2300 MHz
CPU5:	  5.0%     46 °C      900 MHz
CPU6:	  5.0%     40 °C     2300 MHz
CPU7:	  3.0%     42 °C     2300 MHz

auto-cpufreq version:
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CalledProcessError(128, ['git', 'describe', '--always'])

Python: 3.9.7
psutil package: 5.8.0
platform package: 1.0.8
click package: 8.0.3
distro package 1.6.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.14 %

Total CPU usage: 7.2 %
Total system load: 0.72
Average temp. of all cores: 42.5 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

```
---

AdnanHodzic, 2021-12-26:
> ```
> cat: /etc/default/grub: No such file or directory
> ```

This seems like a specific Pop OS problem which isn't really related to auto-cpufreq or its functionality.

closed 2021-12-26 by AdnanHodzic

ImHereByChance, 2022-02-01:
@DanielAbrante, if you still interested (or if someone have the same issue): Pop_os uses systemd-boot instead of GRUB, so you have to configure it to disable intel_pstate. Path to boot config file in my case looks like `/boot/efi/loader/entries/Pop_OS-current.conf` - may vary depending on your partitions scheme.

dswapnil104, 2022-05-23:
> @DanielAbrante, if you still interested (or if someone have the same issue): Pop_os uses systemd-boot instead of GRUB, so you have to configure it to disable intel_pstate. Path to boot config file [...]

I am stuck at the same stone block, so what is the usage / command to switch from intel_pstate to acpi_cpufreq driver via systemd/kernelstub ?
Any help is appreciated !