AdnanHodzic/auto-cpufreq #336: Fedora 35: Errors while installing with install script.

dolitehgreat97, 2021-12-27:
### Error output:
```
Setting up Python environment

./auto-cpufreq-installer: line 117: yum: command not found

Installing necessary Python packages

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (21.3.1)
Requirement already satisfied: setuptools in /opt/auto-cpufreq/venv/lib/python3.10/site-packages (from -r requirements.txt (line 1)) (57.4.0)
Collecting psutil
  Using cached psutil-5.8.0.tar.gz (470 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: click in /opt/auto-cpufreq/venv/lib/python3.10/site-packages/click-8.0.3-py3.10.egg (from -r requirements.txt (line 3)) (8.0.3)
Requirement already satisfied: distro in /opt/auto-cpufreq/venv/lib/python3.10/site-packages/distro-1.6.0-py3.10.egg (from -r requirements.txt (line 4)) (1.6.0)
Using legacy 'setup.py install' for psutil, since package 'wheel' is not installed.
Installing collected packages: psutil
    Running setup.py install for psutil ... error
    ERROR: Command errored out with exit status 1:
     command: /opt/auto-cpufreq/venv/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-4aqnagoq/psutil_d21ff1542b5c41b4b0a78b374d5e674d/setup.py'"'"'; __file__='"'"'/tmp/pip-install-4aqnagoq/psutil_d21ff1542b5c41b4b0a78b374d5e674d/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-3l71nx_9/install-record.txt --single-version-externally-managed --compile --install-headers /opt/auto-cpufreq/venv/include/site/python3.10/psutil
         cwd: /tmp/pip-install-4aqnagoq/psutil_d21ff1542b5c41b4b0a78b374d5e674d/
    Complete output (45 lines):
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.10
    creating build/lib.linux-x86_64-3.10/psutil
    copying psutil/_psosx.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_pssunos.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_pslinux.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_psaix.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/__init__.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_psposix.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_common.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_compat.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_psbsd.py -> build/lib.linux-x86_64-3.10/psutil
    copying psutil/_pswindows.py -> build/lib.linux-x86_64-3.10/psutil
    creating build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/runner.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_contracts.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/__main__.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_linux.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_process.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_osx.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/__init__.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_sunos.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_misc.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_windows.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_posix.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_unicode.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_memleaks.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_connections.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_bsd.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_testutils.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_system.py -> build/lib.linux-x86_64-3.10/psutil/tests
    copying psutil/tests/test_aix.py -> build/lib.linux-x86_64-3.10/psutil/tests
    running build_ext
    building 'psutil._psutil_linux' extension
    creating build/temp.linux-x86_64-3.10
    creating build/temp.linux-x86_64-3.10/psutil
    gcc -Wno-unused-result -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -DPSUTIL_POSIX=1 -DPSUTIL_SIZEOF_PID_T=4 -DPSUTIL_VERSION=580 -DPSUTIL_LINUX=1 -I/opt/auto-cpufreq/venv/include -I/usr/include/python3.10 -c psutil/_psutil_common.c -o build/temp.linux-x86_64-3.10/psutil/_psutil_common.o
    psutil/_psutil_common.c:9:10: fatal error: Python.h: No such file or directory
        9 | #include <Python.h>
          |          ^~~~~~~~~~
    compilation terminated.
    error: command '/usr/bin/gcc' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /opt/auto-cpufreq/venv/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-4aqnagoq/psutil_d21ff1542b5c41b4b0a78b374d5e674d/setup.py'"'"'; __file__='"'"'/tmp/pip-install-4aqnagoq/psutil_d21ff1542b5c41b4b0a78b374d5e674d/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-3l71nx_9/install-record.txt --single-version-externally-managed --compile --install-headers /opt/auto-cpufreq/venv/include/site/python3.10/psutil Check the logs for full command output.

[... 67 lines cut ...]
Cannot run install or debug commands due to this error message. I'm running:
* Fedora 35
* Kernel: 5.15.10-200.fc35.x86_64
* AMD Ryzen 7 PRO 3700U w/ Radeon Vega Mobile Gfx (8) @ 2.300GHz 

```Traceback (most recent call last):
  File "/opt/auto-cpufreq/venv/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.0', 'auto-cpufreq')
  File "/opt/auto-cpufreq/venv/lib64/python3.10/site-packages/pkg_resources/__init__.py", line 3243, in <module>
    def _initialize_master_working_set():
  File "/opt/auto-cpufreq/venv/lib64/python3.10/site-packages/pkg_resources/__init__.py", line 3226, in _call_aside
    f(*args, **kwargs)
  File "/opt/auto-cpufreq/venv/lib64/python3.10/site-packages/pkg_resources/__init__.py", line 3255, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/opt/auto-cpufreq/venv/lib64/python3.10/site-packages/pkg_resources/__init__.py", line 568, in _build_master
    ws.require(__requires__)
  File "/opt/auto-cpufreq/venv/lib64/python3.10/site-packages/pkg_resources/__init__.py", line 886, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/opt/auto-cpufreq/venv/lib64/python3.10/site-packages/pkg_resources/__init__.py", line 772, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'psutil' distribution was not found and is required by auto-cpufreq

```
---
Note: I also tried the snap method and encountered a similar issue. I can create a separate issue for that if needed.

dolitehgreat97, 2021-12-27:
I've seem to have found the error, or at least a solution to this problem: installing python3-devel has provided psutils and the install script works without errors. Which is odd seeing how I have the stand alone package installed.

bobslept, 2021-12-27:
We use `yum` in our installer script. This command is not found your system. See the error message:
`./auto-cpufreq-installer: line 117: yum: command not found`

This command is normally available on a fresh install of Fedora 35. And is just a symlink to dnf.

We do a `yum install python-devel` in the installer script which installs:

```
[beep@fedora ~]$ sudo yum install python-devel
Last metadata expiration check: 0:50:41 ago on Mon 27 Dec 2021 02:22:18 PM CET.
Package python3-devel-3.10.1-1.fc35.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
```

dolitehgreat97, 2021-12-28:
Hmm, I don't have a fresh install, been upgrading since F33, so I'm unsure why I don't have that symlink.

closed 2021-12-28 by dolitehgreat97