AdnanHodzic/auto-cpufreq #337: Cannot start auto-cpufreq via openrc.

FrostKnight, 2021-12-27:
Bug INFO: 

doas rc-service auto-cpufreq start -v
 * Executing: /usr/libexec/rc/sh/openrc-run.sh /usr/libexec/rc/sh/openrc-run.sh /etc/init.d/auto-cpufreq start
 * Starting auto-cpufreq ...
 * supervise-daemon: --pidfile must be specified
 * failed to start auto-cpufreq                                           [ !! ]
 * ERROR: auto-cpufreq failed to start
 

System INFO:

Hyperbola Gnu/Linux-Libre 0.4
It uses a fork of openrc 0.28 which is called hyperrc 0.28-4
https://git.hyperbola.info:50100/packages/core.git/tree/hyperrc

I should also mention, whether I changed the prefix on everything to /usr

or left it alone, same error

Although, Hyperbola usually uses FHS

https://wiki.hyperbola.info/doku.php?id=en:main:fhs

They forked the old version of openrc, due to the newer ones being more bloated.

The odd thing though, is that once I install it, though rc-service cannot start it, I noticed that   the below command

auto-cpufreq --live 

It appears still, hope this helps you in fixing this.

bobslept, 2021-12-28:
The openrc file we made, is setup for using the python virtual environment. Which is used only when using the `auto-cpufreq-installer`. And we point it to the auto-freq wrapper.

So if you installed it via another way, for example via PKGBUILD. You need to edit the openrc file yourself. Just like they did with the systemd service file on the AUR package. They made it point to `/usr/bin/auto-cpufreq --daemon`. 

Hope this fixes your problem.

FrostKnight, 2022-02-06:
Almost forgot, to mention, it seems to work now, the devs for Hyperbola made a few changes due to it using FHS instead of FSH.

https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard

Anywho, I hope you don't mind, but this seems to be solved albeit indirectly. 

;)

closed 2022-02-06 by FrostKnight