AdnanHodzic/auto-cpufreq #339: My fan was running a lot

Pound-Hash, 2022-01-01:
I noticed my fan was running a lot. So I uninstalled `auto-cpufreq` and tried [power-profiles-daemon](https://gitlab.freedesktop.org/hadess/power-profiles-daemon#power-profiles-daemon) instead. That app comes with the command `powerprofilesctl`, which prints:

```
  performance:
    Driver:     intel_pstate
    Degraded:   yes (high-operating-temperature)
```

If the **performance** option is degraded, I would think it preferable to not run it. Now that it's not running via `auto-cpufreq`, my fan doesn't run so much.

- OS: Arch Linux x86_64
- Host: Latitude 7480
- Kernel: 5.15.12-arch1-1
- CPU: Intel i5-6300U (4) @ 2.4GHz

AdnanHodzic, 2022-01-01:
Consider making siniliar configuration as part of [auto-cpufreq config file](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq) or even [changing the driver](https://github.com/AdnanHodzic/auto-cpufreq/#troubleshooting).

Pound-Hash, 2022-01-01:
I had already tried the alternative driver. I'll try altering the configuration. Thanks.

Alan

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

molitona, 2022-01-07:
@AdnanHodzic 
My fan was running a lot too also...

AdnanHodzic, 2022-01-08:
@Pound-Hash changes regarding setting `balanced` mode with `power-profiles-daemon` were merged last night as part of #349, hence can you try pulling latest changes from `master` and see if you still have the same problem? Also make sure to include `auto-cpufreq --debug`

@molitona I think problem in your case is that your charging states are not detected properly, originally introduced with #308, hence @bobslept could you please take a look at this? Thank you

AdnanHodzic, 2022-01-09:
I think [v1.9.1 release](https://github.com/AdnanHodzic/auto-cpufreq/releases) should fix this issue.

molitona, 2022-01-22:
@Pound-Hash Does the problem is fixed for you with the latest version ?

Pound-Hash, 2022-01-22:
> @Pound-Hash Does the problem is fixed for you with the latest version ?

I haven't tried it yet. The problem was resolved a few weeks ago when I simply disabled performance mode.

molitona, 2022-01-22:
Are you planning to test it ?

Pound-Hash, 2022-01-22:
> Are you planning to test it ?

Sure, but I'm sick and I have a lot of personal issues to tend to. I won't get around to playing with Linux for a couple weeks.

Pound-Hash, 2022-01-25:
I misremembered what I did earlier. I didn't actually disable performance mode; instead, I disabled **turbo**. I just installed the latest version. Testing now with turbo reset to auto.

jeankhawand, 2022-01-27:
I am facing the same issue fan running at full speed while CPU usage ~2.7%
```
-------------------------------------------------------------------------------

Linux distro: Ubuntu 20.04 focal
Linux kernel: 5.13.0-27-generic
Processor: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
Cores: 12
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2601 MHz
CPU min frequency: 800 MHz

Core	Usage	Temperature	Frequency
CPU0:	  8.0%     48 °C     2601 MHz
CPU1:	  5.0%     49 °C     2601 MHz
CPU2:	  2.0%     47 °C     2601 MHz
CPU3:	  2.0%     47 °C     2600 MHz
CPU4:	 18.4%     47 °C     2601 MHz
CPU5:	 18.0%     53 °C     2601 MHz
CPU6:	  3.0%     48 °C     2601 MHz
CPU7:	  5.0%     49 °C     2601 MHz
CPU8:	  2.0%     47 °C     2601 MHz
CPU9:	  7.9%     47 °C     2601 MHz
CPU10:	  4.0%     47 °C     2601 MHz
CPU11:	  8.1%     53 °C     2601 MHz

---------------------------- CPU frequency scaling ----------------------------

Battery is: charging

Setting to use: "performance" governor

Total CPU usage: 2.7 %
Total system load: 0.36
Average temp. of all cores: 48.5 °C

Load optimal
setting turbo boost: off
```

Pound-Hash, 2022-02-01:
After testing for one week, the problem appears resolved - the fan doesn't run so much.

Current config:
```
### Modified ###

[charger]
# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
governor = performance

# always, auto, never
turbo = auto

[battery]
governor = powersave
turbo = auto
```

jeankhawand, 2022-02-02:
> After testing for one week, the problem appears resolved - the fan doesn't run so much.
>
> Current config:
> ```
> ### Modified ###
>
> [charger]
> # cat [...]

@Pound-Hash which driver did you use ?

Pound-Hash, 2022-02-02:
> @Pound-Hash which driver did you use ?

`intel_pstate`

AdnanHodzic, 2022-02-20:
Just [released v1.9.2](https://github.com/AdnanHodzic/auto-cpufreq/releases) which adds ability to change power governor used by default with auto-cpufreq from `balanced` to `performance` [using power_helper.py script](https://github.com/AdnanHodzic/auto-cpufreq/#1-power_helperpy-script). Curious to hear if you still have same problem if you switch to `performance` governor?

Pound-Hash, 2022-02-22:
My fan accelerates when building from source if performance is allowed. If I enable turbo, it accelerates more; if disabled, it accelerates less. Conversely, if I disable performance and turbo, then the fan will never accelerate even when building from source.

AdnanHodzic, 2022-02-27:
> My fan accelerates when building from source if performance is allowed. If I enable turbo, it accelerates more; if disabled, it accelerates less. Conversely, if I disable performance and turbo, then the fan will never accelerate even when building from source.

What you explained here is exactly how system should act under these circumstances, hence could you please elaborate on what exactly is the problem?

Pound-Hash, 2022-02-27:
> > My fan accelerates when building from source if performance is allowed. If I enable turbo, it accelerates more; if disabled, it accelerates less. Conversely, if I disable performance and turbo, then the fan will never accelerate even when building from source.
> 
> What you explained here is exactly how system should act under these circumstances, hence could you please elaborate on what exactly is the problem?
> 
> Also make sure you don't have `laptop-mode-tools` installed as mentioned in #365

My question was stated in the OP.

`powerprofilesctl` says the performance option is degraded, and seems to also say that it causes high operating temperature. So the question was, should we be running it at all.

AdnanHodzic, 2022-06-10:
You should and [should instead used "balanced"](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq), closing the issue please feel free to re-open if you're still facing the same issue.

closed 2022-06-10 by AdnanHodzic

Pound-Hash, 2022-06-11:
> You should and [should instead used "balanced"](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq), closing the issue please feel free to re-open if you're still facing the same issue.

I should and should? I'm not sure I understand.

AdnanHodzic, 2022-06-11:
Sorry, made a typo, but I also misunderstood your last statement. You said, this works config for you:

> After testing for one week, the problem appears resolved - the fan doesn't run so much.
>
> Current config:
>
> ```
> ### Modified ###
>
> [charger]
> # cat [...]

Which is good, but then you stated you're facing this issue:

>My fan accelerates when building from source if performance is allowed. If I enable turbo, it accelerates more; if disabled, it >accelerates less. Conversely, if I disable performance and turbo, then the fan will never accelerate even when building from >source.

What I was trying to say is you should [switch to auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq#configuring-auto-cpufreq) `performance` mode. I just updated this section of README to make it more elaborate on how to do so. Namely, I had this same problem as (#361) and I fixed it by introducing this mode.

Hope it's all clear now :)