AdnanHodzic/auto-cpufreq #341: [Request] Add update command

ariasmn, 2022-01-03:
Currently, as discussed in #340 , for those that installed auto-cpufreq using the `auto-cpufreq-installer`, the update needs to be done manually, by downloading the new release and installing again.

Implemeting a new command, like `auto-cpufreq --update` to automatically do so would be a nice feature.

AdnanHodzic, 2022-11-11:
Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

cosmos1721, 2023-06-14:
hey @AdnanHodzic 
can i work on this issue?

AdnanHodzic, 2023-06-14:
@cosmos1721 please do, and once you have something please create a PR as I refered to above.

AdnanHodzic, 2023-07-15:
Thanks to @cosmos1721 contribution, `--update` feature added as part of #535 has already been merged to `master` branch and will be part of upcoming v1.9.9 release. 

Hence closing the issue.

closed 2023-07-15 by AdnanHodzic