AdnanHodzic/auto-cpufreq #344: Current version not showing when installed via source code

cnske, 2022-01-04:
### Error output:
```
auto-cpufreq version:
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CalledProcessError(128, ['git', 'describe', '--always'])
```
#### Additional info
The current version is not showing up. I pulled the code via git into my `/home` and installed it from there. Is there an option to tell the current installed version where to find the git repo and get the info regarding the version. Moreover, I wonder what would be the proper way to update the software and if it would be better to put the git repo into `/opt/auto-cpufreq`.

---

### System Information
```
Linux distro: Debian GNU/Linux 11 bullseye
Linux kernel: 5.10.0-9-amd64
Processor: Intel(R) Core(TM) i9-8950HK CPU @ 2.90GHz
Cores: 12
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2900 MHz
CPU min frequency: 800 MHz

Core	Usage	Temperature	Frequency
CPU0:	  1.1%     40 °C      900 MHz
CPU1:	  1.0%     37 °C      900 MHz
CPU2:	  2.0%     38 °C      900 MHz
CPU3:	  8.8%     37 °C      900 MHz
CPU4:	  0.0%     38 °C      900 MHz
CPU5:	  2.9%     38 °C      900 MHz
CPU6:	  1.0%     40 °C      900 MHz
CPU7:	  0.0%     37 °C      900 MHz
CPU8:	  1.0%     38 °C      900 MHz
CPU9:	  1.0%     37 °C      900 MHz
CPU10:	  3.0%     38 °C      900 MHz
CPU11:	  0.0%     38 °C      900 MHz

auto-cpufreq version:
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CalledProcessError(128, ['git', 'describe', '--always'])

Python: 3.9.2
psutil package: 5.9.0
platform package: 1.0.8
click package: 8.0.3
distro package 1.6.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.07 %

Total CPU usage: 3.0 %
Total system load: 0.92
Average temp. of all cores: 38.0 °C 

Currently using: powersave governor
Currently turbo boost is: off
```
---

ariasmn, 2022-01-04:
Can confirm. I bumped into the same error while looking into #341 . This issue should be fixed before trying to implement the update feature in my opinion.

closed 2022-01-09 by AdnanHodzic

AdnanHodzic, 2022-01-09:
Issue permanently resolved as part of [1.9.1 release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

PR #347 (2022-01-09): Fixed the --version command when installed from source
Closes #344

The command only worked when running it inside the git repository.

Now, it works from everywhere, and the version is constructed using the latest git tag and the SHA (8 first symbols) of the latest commit, which may help with further debugging in future issues (since when installing from source you don't install a fixed version).

To do so, I have included [this setuptools plugin](https://github.com/dolfinus/setuptools-git-versioning). The template for the versioning [can be changed](https://setuptools-git-versioning.readthed [... cut]