AdnanHodzic/auto-cpufreq #360: Can't uninstall auto-cpufreq when installed through installer. Also no file /etc/auto-cpufreq.conf present to edit.

surf3273, 2022-02-06:
### Error output:
```text
Add/paste error output
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

#### Without providing requested information your issue might be ignored!

---

AdnanHodzic, 2022-02-06:
Closing due to user not following instructions. 

If you run installer you can follow on screen instructions to install/remove auto-cpufreq: `sudo ./auto-cpufreq-installer`. [auto-cpufreq config file is not there by default](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq), you need to add and then you'll have have something to edit.

closed 2022-02-06 by AdnanHodzic

surf3273, 2022-02-07:
Thank you so much sir and excuse me for such a juvenile issue as I am very new to Linux and its framework.

niamotullah, 2024-04-18:
> Closing due to user not following instructions.
>
> If you run installer you can follow on screen instructions to install/remove auto-cpufreq: `sudo ./auto-cpufreq-installer`. [auto-cpufreq config file [...]

Please add this to the README

AdnanHodzic, 2024-04-20:
> Please add this to the README

@niamotullah 

If there are any changes that could be made please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).