AdnanHodzic/auto-cpufreq #365: Prevent mouse and sound-card from entering power-saving mode

Greeenstone, 2022-02-20:
Hey,   
Due to some sound issues, I had to disable powersaving mode for my sound card (i.e. set `options snd-hda-intel power_save=0` in `/etc/modprobe.d/sound.conf`)
In battery it appears that the soundcard enters powersaving again. 
In addition my usb-mouse goes to 'sleep' (after 3sec I need to reactivate it by clicking first)

### System information:
my `auto-cpufreq --debug` output:
```
-------------------------------------------------------------------------------

Linux distro: Arch Linux  
Linux kernel: 5.15.24-1-lts
Processor: Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1600 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  2.0%     37 °C     1800 MHz
CPU1:	  1.0%     38 °C     1800 MHz
CPU2:	  0.0%     37 °C     1800 MHz
CPU3:	  0.0%     38 °C     1121 MHz
CPU4:	  1.0%     37 °C     1800 MHz
CPU5:	  1.0%     38 °C     1800 MHz
CPU6:	  4.0%     37 °C     1800 MHz
CPU7:	  2.0%     38 °C     1800 MHz

auto-cpufreq version: Version         : 1.9.1-3

Python: 3.10.2
psutil package: 5.9.0
platform package: 1.0.8
click package: 8.0.3
distro package: 1.6.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.28 %

Total CPU usage: 1.4 %
Total system load: 0.74072265625
Average temp. of all cores: 37.5 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```

#### Request Information
Is setting to sleep the soundcard and usb-mouse a **bug** or a **feature**?  
Is there an option to prevent it from doing so?

Furthermore, is it desired that 7/8 cores are full frequency even though there is very little CPU-usage?

Cheers,
---

AdnanHodzic, 2022-02-21:
Hey, if I was you I would configure your sound card/mouse not to go to sleep when in `powersave`, it should be simple as that, really has nothing to do with auto-cpufreq.

If you're not fully happy how auto-cpufreq does things automatically you can [always configure it more to your own liking](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq).

Greeenstone, 2022-02-22:
After some digging, I found that my issue was due to the app "laptop-mode-tools". Thus, I apologize for the confusion.
Thanks anyway for the great piece of software and maintenance :)

closed 2022-02-22 by Greeenstone

AdnanHodzic, 2022-02-27:
> 

Thanks for coming back to me with this, auto-cpufreq [automatically notifies user if TLP is detected](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto_cpufreq/power_helper.py#L52) which [causes issues when running together with auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq#why-do-i-need-auto-cpufreq), hence same type of check should be made for `laptop-mode-tools`

I always encourage people to [make code contributions](https://github.com/AdnanHodzic/auto-cpufreq#code-contribution) if they want to give it a try themselves in resolving this issues for good.