AdnanHodzic/auto-cpufreq #372: sh: 1: snap: Permission denied

meetdilip, 2022-03-05:
Auto CPU Frequency was working fine for around 6 months for me. Suddenly a couple of days before, I got this permission denied error even after using `sudo`. I was using `sudo auto-cpufreq` fine for months and suddenly this error. It would be nice if you can do anything.

### Error output:
sh: 1: snap: Permission denied
Add/paste error output
```
---

### System information:

HP Pavilion Laptop

AMD Ryzen 5600H + Nvidia 1650 4GB + 8 GB RAM

Add/paste output of:

```
auto-cpufreq --debug
```

$ sudo auto-cpufreq
[sudo] password for deep: 
sh: 1: snap: Permission denied

-------------------------------- auto-cpufreq ---------------------------------

Automatic CPU speed & power optimizer for Linux

Example usage:
auto-cpufreq --monitor

-----

sh: 1: snap: Permission denied
Usage: auto-cpufreq [OPTIONS]

Options:
  --monitor             Monitor and see suggestions for CPU optimizations
  --live                Monitor and make (temp.) suggested CPU optimizations
  --install / --remove  Install/remove daemon for (permanent) automatic CPU
                        optimizations
  --stats               View live stats of CPU optimizations made by daemon
  --config TEXT         Use config file at defined path
  --debug               Show debug info (include when submitting bugs)
  --version             Show currently installed version
  --donate              Support the project
  --help                Show this message and exit.

-------------------------------------------------------------------------------

#### Without providing requested information your issue might be ignored!

---

meetdilip, 2022-03-07:
Hi, I made a Snapcraft forum topic. They can replicate the issue. Please check

https://forum.snapcraft.io/t/sh-1-snap-permission-denied/29023

AdnanHodzic, 2022-03-08:
Hi @meetdilip thanks for looking into this and trying to help, I've also replied on the topic you created. 

This issue is closely related (if not a duplicate) to #368. There I also created a new branch and did [some work on which could potentially fix the issue](https://github.com/AdnanHodzic/auto-cpufreq/issues/368#issuecomment-1055052649), and if you have time or knowledge or time to look into this problem [please do](https://github.com/AdnanHodzic/auto-cpufreq#code-contribution) and I'll be glad to give you credit as part of new release.

AdnanHodzic, 2022-03-08:
> I am not a coder though. 

There are other ways to [support the project](https://github.com/AdnanHodzic/auto-cpufreq#financial-donation) which no one cares about as long as others are coding :)

> Auto CPU Frequency was working fine for me since 21.04. Now I am using 21.10. This issue happened somewhere last week.

This has nothing to do with Ubuntu version, [as I mentioned](https://github.com/AdnanHodzic/auto-cpufreq/issues/368#issuecomment-1055052649) in #368 problem was introduced as part of [1.9.2 release](https://github.com/AdnanHodzic/auto-cpufreq/releases) which was released around that time.

> Is there any method to use the previous version as it was working flawlessly?

You should ask Snap folks about this, which I see you did. I just pushed v1.9.3 which is in development to Snap beta channel which should address the problem [mentioned on Snap forum](https://forum.snapcraft.io/t/sh-1-snap-permission-denied/29023/10?u=ahodzic).

Please install it by running: `sudo snap install auto-cpufreq --beta` and let me know if this fixes your issue.

meetdilip, 2022-03-08:
This is what I got. Does it mean it is ok now? Thanks

> $ sudo auto-cpufreq
> 
> -------------------------------- auto-cpufreq ---------------------------------
> 
> Automatic CPU speed & power optimizer for Linux
> 
> Example usage:
> auto-cpufreq --monitor
> 
> -----
> 
> Usage: auto-cpufreq [OPTIONS]
> 
> Options:
>   --monitor             Monitor and see suggestions for CPU optimizations
>   --live                Monitor and make (temp.) suggested CPU optimizations
>   --install / --remove  Install/remove daemon for (permanent) automatic CPU
>                         optimizations
>   --stats               View live stats of CPU optimizations made by daemon
>   --config TEXT         Use config file at defined path
>   --debug               Show debug info (include when submitting bugs)
>   --version             Show currently installed version
>   --donate              Support the project
>   --help                Show this message and exit.

meetdilip, 2022-03-08:
Got this for --live
[image]

AdnanHodzic, 2022-03-08:
LGTM, run `sudo auto-cpufreq --install` and have it run everything automatically on your behalf. 

Seems that latest (unreleased) changes fix the problem. After I verify problem has also been fixed on non snap installs (#386), I'll make a new release and after I done that I'll close this issue. As part of that I'll also inform you and instruct how to switch back to stable channel.

AdnanHodzic, 2022-03-09:
Great, looks like you have successful install. Warning messages are something optional you can do if you're facing issues, but nothing related to your original problem.

Can I also see output of `auto-cpufreq --stats`?

AdnanHodzic, 2022-03-09:
LGTM, problem fixed, closing the issue. As I said, once the new release has been made I'll update this issue with how to switch back to stable release.

closed 2022-03-09 by AdnanHodzic

AdnanHodzic, 2022-03-10:
[v1.9.3 has been released](https://github.com/AdnanHodzic/auto-cpufreq/releases) which resolves this issue.

@meetdilip to switch back to "stable" channel of this snap package, please run `sudo snap switch --channel=stable auto-cpufreq`

meetdilip, 2022-03-10:
Hi, Thanks

$ sudo snap switch --channel=stable auto-cpufreq
[sudo] password for deep: 
"auto-cpufreq" switched to the "latest/stable" channel

meetdilip, 2022-03-10:
```
$ auto-cpufreq --version

-------------------------------------------------------------------------------

Linux distro: Ubuntu 21.10 (Impish Indri)
Linux kernel: 5.13.0-28-generic
auto-cpufreq version: (Snap) 1.9.3
```