AdnanHodzic/auto-cpufreq #376: Use /usr/local/bin/ instead of /usr/bin/

gimbles, 2022-03-08:
Using /usr/local/bin/ will make this Fedora silverblue compatible. Currently I am unable to install it on my laptop.

AdnanHodzic, 2022-03-08:
Hi @gimbles glad you made it work, could you please [make a code contribution](https://github.com/AdnanHodzic/auto-cpufreq#code-contribution) so we have auto-cpufreq also working on Silverblue? I think it could be simple as just adding case for silverblue as part of [auto-cpufreq-install.sh](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/scripts/auto-cpufreq-install.sh)

Thanks!

gimbles, 2022-03-15:
> Hi @gimbles glad you made it work, could you please [make a code contribution](https://github.com/AdnanHodzic/auto-cpufreq#code-contribution) so we have auto-cpufreq also working on Silverblue? I [...]

I..was not able to make it work, that's kinda the point of the issue :stuck_out_tongue:

gimbles, 2022-03-15:
and I'm currently on Fedora workstation for the meanwhile, don't think I'll be able to help for silverblue now :(

gimbles, 2022-05-15:
Update -> 

https://github.com/gimbles/auto-cpufreq-silverblue

closed 2022-05-15 by gimbles