AdnanHodzic/auto-cpufreq #381: /usr/bin/cpufreqctl.auto-cpufreq is constantly being written to

popsUlfr, 2022-03-28:
Hi,

while using auto-cpufreq I noticed that it was writing to disk on every loop.

By using strace 
```
sudo strace -f -e trace=file -p <auto-cpufreq-pid> |& grep -e 'open.*WR' -e 'open.*RW'
```
```
openat(AT_FDCWD, "/usr/bin/cpufreqctl.auto-cpufreq", O_WRONLY|O_CREAT|O_TRUNC|O_CLOEXEC, 0666) = 5
...
```

I saw it was `/usr/bin/cpufreqctl.auto-cpufreq` getting constantly written to which is weird.

It seems to be this part:
https://github.com/AdnanHodzic/auto-cpufreq/blob/ae1f53c737d9f6462e980bc1614ee6ca1f6b36cd/auto_cpufreq/core.py#L286-L299

https://github.com/AdnanHodzic/auto-cpufreq/blob/ae1f53c737d9f6462e980bc1614ee6ca1f6b36cd/auto_cpufreq/core.py#L299

I'm not entirely sure what should happen here but I don't think `/usr/bin/cpufreqctl.auto-cpufreq` is supposed to be copied in both cases.

---

### System information:

```

-------------------------------------------------------------------------------

Linux distro: Arch Linux rolling n/a
Linux kernel: 5.16.16-zen1-1-zen
Processor: AMD Ryzen 9 5900X 12-Core Processor
Cores: 24
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3700 MHz
CPU min frequency: 2200 MHz

Core    Usage   Temperature     Frequency
CPU0:     2.0%     44 °C     3700 MHz
CPU1:     1.0%     44 °C     3700 MHz
CPU2:     0.0%     44 °C     3700 MHz
CPU3:     0.0%     44 °C     3700 MHz
CPU4:     0.0%     44 °C     3700 MHz
CPU5:     0.0%     44 °C     3700 MHz
CPU6:     0.0%     44 °C     3700 MHz
CPU7:     0.0%     44 °C     3700 MHz
CPU8:     0.0%     44 °C     3700 MHz
CPU9:     0.0%     44 °C     3700 MHz
CPU10:    1.0%     44 °C     3700 MHz
CPU11:    0.0%     44 °C     3700 MHz
CPU12:    0.0%     44 °C     3700 MHz
CPU13:    1.0%     44 °C     3700 MHz
CPU14:    0.0%     44 °C     3700 MHz
CPU15:    0.0%     44 °C     3700 MHz
CPU16:    0.0%     44 °C     3700 MHz
CPU17:    0.0%     44 °C     3700 MHz
CPU18:    0.0%     44 °C     3700 MHz
CPU19:    0.0%     44 °C     3700 MHz
CPU20:    0.0%     44 °C     3215 MHz
CPU21:    0.0%     44 °C     3700 MHz
CPU22:    1.0%     44 °C     3700 MHz
CPU23:    0.0%     44 °C     3700 MHz

auto-cpufreq version: Version         : 1.9.3-1

Python: 3.10.3
psutil package: 5.9.0
platform package: 1.0.8
click package: 8.0.4
distro package: 1.7.0

Computer type: Desktop
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.08 %

Total CPU usage: 1.0 %
Total system load: 0.333984375
Average temp. of all cores: 43.5 °C 

Currently using: performance governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```

AdnanHodzic, 2022-04-25:
I always advise people to [make their own code contributions](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) if they have a good idea how certain things should be done.

AdnanHodzic, 2022-08-15:
Issue has been fixed as part of #420, which is now part of `master` branch and will be released as part of upcoming 1.9.6 release.

If problem persist, please re-open the issue.

closed 2022-08-15 by AdnanHodzic