AdnanHodzic/auto-cpufreq #382: Not scaling up

Yury-MonZon, 2022-04-03:
My CPU is kept on minimum frequency no matter how heavy I load it. 
I've tried stress -c 4 in addition to multiple youtube videos playing. It shows all cores are maxed to 100% load, but the frequency stays at the minimum level.

--debug output is done right after I closed stress program.
### System information:
I'm using macbook air 11". intel_pstate is disabled. TLP is not installed. In macos cpu freq scaling works fine, TLP was doing fine, but a bit hot. 
```
sudo auto-cpufreq --debug

-------------------------------------------------------------------------------

Linux distro: Linux Mint 20.3 Una
Linux kernel: 5.4.0-107-generic
Processor: Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz
Cores: 4
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2301 MHz
CPU min frequency: 759 MHz

Core	Usage	Temperature	Frequency
CPU0:	 12.5%     66 °C      800 MHz
CPU1:	  1.0%     65 °C      800 MHz
CPU2:	  4.0%     66 °C      800 MHz
CPU3:	  0.0%     65 °C      800 MHz

auto-cpufreq version: 1.9.3 (git: ae1f53c7)

Python: 3.8.10
psutil package: 5.9.0
platform package: 1.0.8
click package: 8.1.2
distro package: 1.7.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.26 %

Total CPU usage: 7.2 %
Total system load: 3.76
Average temp. of all cores: 65.5 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

```

#### Without providing requested information your issue might be ignored!

---

If installation failed with [auto-cpufreq-installer ](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer)have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
I don't have any errors installing from git.

Yury-MonZon, 2022-04-03:
```
-------------------------------------------------------------------------------

Linux distro: Linux Mint 20.3 Una
Linux kernel: 5.4.0-107-generic
Processor: Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz
Cores: 4
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2301 MHz
CPU min frequency: 759 MHz

Core	Usage	Temperature	Frequency
CPU0:	100.0%     65 °C      800 MHz
CPU1:	100.0%     66 °C      800 MHz
CPU2:	100.0%     65 °C      800 MHz
CPU3:	100.0%     66 °C      800 MHz

auto-cpufreq version: 1.9.3 (git: ae1f53c7)

Python: 3.8.10
psutil package: 5.9.0
platform package: 1.0.8
click package: 8.1.2
distro package: 1.7.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.26 %

Total CPU usage: 99.7 %
Total system load: 3.61
Average temp. of all cores: 65.5 °C 

Currently using: powersave governor
Currently turbo boost is: on

-------------------------------------------------------------------------------

```

AdnanHodzic, 2022-04-25:
Have you tried using [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq) to use "performance"? Does that change anything?

Nova1545, 2022-05-12:
I am also having a very similar issues, the CPU never leaves power-save and nevr clocks up, even using the suggestion above

AdnanHodzic, 2022-06-10:
Can someone confirm if this is related to: #398

MarwenDallel, 2022-06-25:
I have exactly the same issue on a different CPU/kernel version. The CPU frequency only reaches 900MHz.   
The problem with switching to performance is that it locks my CPU frequency to a minimum of 1600MHz no matter the configuration parameters I specify. In my case, performance doesn't scale down (at least not any lower than the base freq) and powersave doesn't scale up.

Debug output (under load):
```
-------------------------------------------------------------------------------

Linux distro: Manjaro Linux 21.3.1 Ruah
Linux kernel: 5.15.49-1-MANJARO
Processor: Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz
Cores: 8
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1600 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0:	100.0%     47 °C      900 MHz
CPU1:	100.0%     46 °C      900 MHz
CPU2:	100.0%     45 °C      900 MHz
CPU3:	100.0%     45 °C      900 MHz
CPU4:	100.0%     47 °C      900 MHz
CPU5:	100.0%     46 °C      900 MHz
CPU6:	100.0%     45 °C      900 MHz
CPU7:	100.0%     45 °C      900 MHz

auto-cpufreq version: Version         : 1.9.4-1

Python: 3.10.5
psutil package: 5.9.0
platform package: 1.0.8
click package: 8.1.3
distro package: 1.7.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.33 %

Total CPU usage: 99.8 %
Total system load: 7.15
Average temp. of all cores: 45.75 °C 

Currently using: powersave governor
Currently turbo boost is: on

-------------------------------------------------------------------------------
```

AdnanHodzic, 2022-11-11:
If there are any changes that could be made please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

In meantime closing the issue due to inactivity.

closed 2022-11-11 by AdnanHodzic