AdnanHodzic/auto-cpufreq #385: [ Feature Request ] Temporarily change cpu governor

renzhamin, 2022-04-09:
Is there any way to change cpu governor on the fly without editing /etc/auto-cpufreq.conf ?
It would be fantastic to be able to change the governor temporarily, such as for current boot, regardless of whether device is running on battery or ac, for example:

```bash
auto-cpufreq --cpu-governor conservative # effective until next boot or restart of the service
auto-cpufreq --cpu-governor auto # usual setting respecting battery/ac mode
```

This can create other opportunities such as changing governor when certain application or a set of applications is found running

AdnanHodzic, 2022-04-25:
I always advise people to [make their own code contributions](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) if they have a good idea how certain things should be done.

AdnanHodzic, 2022-11-11:
Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).