AdnanHodzic/auto-cpufreq #387: FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors'

ACNAryan5498, 2022-04-20:
### Error output:

If installation failed with [auto-cpufreq-installer ](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer)have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?

I tried to install auto-cpufreq in both ways, using `auto-cpufreq-installer` and `snap`. Neither of ways work out for me.

First I try using `snap` package, and it gives me this error : 

`mint@mintPC:~$ sudo snap run auto-cpufreq --monitor

Note: You can quit monitor mode by pressing "ctrl+c"

----------------------------------- Warning -----------------------------------

Unable to detect state of GNOME Power Profiles daemon service!
This daemon might interfere with auto-cpufreq and should be disabled.

Steps to perform this action using auto-cpufreq: power_helper script:
git clone https://github.com/AdnanHodzic/auto-cpufreq.git
cd auto-cpufreq/auto_cpufreq
python3 power_helper.py --gnome_power_disable

----------------------------------- Warning -----------------------------------

Unable to detect if you are using a TLP service!
This daemon might interfere with auto-cpufreq which can lead to unexpected results.
We strongly encourage you not to use TLP unless you really know what you are doing.

-------------------------------------------------------------------------------

Traceback (most recent call last):
  File "/snap/auto-cpufreq/112/bin/auto-cpufreq", line 226, in <module>
    main()
  File "/snap/auto-cpufreq/112/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/snap/auto-cpufreq/112/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/snap/auto-cpufreq/112/lib/python3.8/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/snap/auto-cpufreq/112/lib/python3.8/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/snap/auto-cpufreq/112/bin/auto-cpufreq", line 97, in main
    gov_check()
  File "/snap/auto-cpufreq/112/lib/python3.8/site-packages/auto_cpufreq/core.py", line 442, in gov_check
    for gov in get_avail_gov():
  File "/snap/auto-cpufreq/112/lib/python3.8/site-packages/auto_cpufreq/core.py", line 259, in get_avail_gov
    return f.read_text().strip().split(" ")
  File "/usr/lib/python3.8/pathlib.py", line 1236, in read_text
    with self.open(mode='r', encoding=encoding, errors=errors) as f:
  File "/usr/lib/python3.8/pathlib.py", line 1222, in open
    return io.open(self, mode, buffering, encoding, errors, newline,
  File "/usr/lib/python3.8/pathlib.py", line 1078, in _opener
    return self._accessor.open(self, flags, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors'`
`sudo auto-cpufreq --monitor`

So I use `sudo snap run auto-cpufreq --debug` : 

`mint@mintPC:~$ sudo snap run auto-cpufreq --debug

-------------------------------------------------------------------------------

Linux distro: Linux Mint 20.3 (Una)
Linux kernel: 5.4.0-91-generic
Processor: Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz
Cores: 4
Architecture: x86_64
Driver: cat: /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver: No such file or directory

------------------------------ Current CPU stats ------------------------------
[... 123 lines cut ...]
`mint@mintPC:~$ neofetch
             ...-:::::-...                 mint@mintPC 
          .-MMMMMMMMMMMMMMM-.              ----------- 
      .-MMMM`..-:::::::-..`MMMM-.          OS: Linux Mint 20.3 x86_64 
    .:MMMM.:MMMMMMMMMMMMMMM:.MMMM:.        Host: KVM/QEMU (Standard PC (i440FX  
   -MMM-M---MMMMMMMMMMMMMMMMMMM.MMM-       Kernel: 5.4.0-91-generic 
 `:MMM:MM`  :MMMM:....::-...-MMMM:MMM:`    Uptime: 1 hour, 12 mins 
 :MMM:MMM`  :MM:`  ``    ``  `:MMM:MMM:    Packages: 1969 (dpkg), 4 (snap) 
.MMM.MMMM`  :MM.  -MM.  .MM-  `MMMM.MMM.   Shell: bash 5.0.17 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   Resolution: 1366x768 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM:MMM:   DE: Xfce 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   WM: Xfwm4 
.MMM.MMMM`  :MM:--:MM:--:MM:  `MMMM.MMM.   WM Theme: Mint-Y-Dark 
 :MMM:MMM-  `-MMMMMMMMMMMM-`  -MMM-MMM:    Theme: Mint-Y-Dark [GTK2], Mint-Y [G 
  :MMM:MMM:`                `:MMM:MMM:     Icons: Mint-Y-Dark [GTK2], Mint-Y [G 
   .MMM.MMMM:--------------:MMMM.MMM.      Terminal: xfce4-terminal 
     '-MMMM.-MMMMMMMMMMMMMMM-.MMMM-'       Terminal Font: Monospace 12 
       '.-MMMM``--:::::--``MMMM-.'         CPU: Intel i5-5200U (4) @ 2.194GHz 
            '-MMMMMMMMMMMMM-'              GPU: 00:02.0 Red Hat, Inc. QXL parav 
               ``-:::::-``                 Memory: 1501MiB / 1987MiB 
`

I don't know what I did wrong, I just follow the steps which is mentioned in `README.md`. It's used to work on my main machine, which I used the installer past 2 months ago. Today I try to install it on my 2nd machine and I didn't work. So I thought of give it try again on the VM. It also give me the same error.

Is there any way to solve bcoz I really love this.

AdnanHodzic, 2022-04-25:
Running auto-cpufreq on a VM doesn't make sense and will not work, instead run auto-cpufreq on host where VM's are running.

ACNAryan5498, 2022-04-30:
OK I wll give it a try on actual machine again

AdnanHodzic, 2022-06-10:
Closed due to inactivity.

closed 2022-06-10 by AdnanHodzic