AdnanHodzic/auto-cpufreq #391: Full Documentation request?

FrostKnight, 2022-05-01:
In my case, its an ivybridge device, with intel me disabled, a hard forked arch based distro using openrc, etc...

But yeah, it would be nice, if you could have all the various example options in one spot, even if they are commented out by default.

I would like to have the option change settings at will, to the way I want it...

Max battery life, with one exception, want to have wifi come back on, when I enabled it, as long as I haven't kill switched it, but when I disable killswitch, it turns on again, etc...

This is only but one example though.

Also, I don't know each specific, function for each available option, etc... 

Can you help me, at some point? 

Thanks in advance, for this lovely project.

anonyco, 2022-05-07:
Here you go: https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto-cpufreq.conf-example

This should close the issue.

AdnanHodzic, 2022-05-07:
Thank you @anonyco. I also believe it's all there, I always encourage folks to [make contributions](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and improve things, if one of you think this can be improved. 

Closing the issue, if problem persists feel free to re-open it.

closed 2022-05-07 by AdnanHodzic

FrostKnight, 2022-05-08:
> Thank you @anonyco. I also believe it's all there, I always encourage folks to [make contributions](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and improve things, if one of you think this can be improved.
> 
> Closing the issue, if problem persists feel free to re-open it.

Actually, are you saying, this is all the documentation you have regarding power settings?

I wanted to know, how to do a few things, such as wifi controls, ethernet controls, etc...

I have a kill switch for wifi/bluetooth, but I kind of wanted, to be able to turn wifi back on, otherwise, it keeps disabling itself...

;)

Btw, no I don't have tlp on my system anymore or laptop-mode-tools, thus, I don't know why that happens...

Its weird.

anonyco, 2022-05-09:
> I wanted to know, how to do a few things, such as wifi controls, ethernet controls, etc...

One of the unix philosophies for software is "do one thing and do one thing well." auto-cupfreq does only cpu frequency tuning and it does this one thing well. You would need to make extensive modifications to the scripts in order to add advanced features like these.

FrostKnight, 2022-05-10:
> One of the unix philosophies for software is "do one thing and do one thing well." auto-cupfreq does only cpu frequency tuning and it does this one thing well. You would need to make extensive [...]

Ah, okay, hmm... 

I didn't know that was how you designed this.

Alrighty, cool, thanks for the info, I didn't know if you were holding off on me...

;) 

Anywho, done and done.

Btw, does this mean you plan to support BSD as well?

I hope so, because ARM64, has no good distro, imo.

I hope to use arm64 more in the future, etc...

OpenBSD is currently the only OS i would trust that currently supports it.

Long story short, you are wise to do this, sorry for bothering you with my lack of knowledge...

AdnanHodzic, 2022-05-14:
I do have plans to implement arm64 support, but I never had enough time to do so far, hence as I mentioned previously if you or anyone else want to give it a go, please [give it a try and contribute](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) :)

FrostKnight, 2022-05-15:
> I do have plans to implement arm64 support, but I never had enough time to do so far, hence as I mentioned previously if you or anyone else want to give it a go, please [give it a try and contribute](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) :)

I would be willing to test, except for two issues:

I don't have an arm64 device, (yet!)
I tend to prefer to test betas if at all.
Aka, alphas usually make me nervous...
Unless I have a virtual machine, etc...
:)
Curious btw, have you heard of something called MNT Reform?
Whenever their pocket version comes out, you better believe, I will want this to support ARM64! Although some probably want that even now for the current version. 

Long story short,  those represent my interest in ARM64, although other options exist, those are most definitely, the most secure/freest out there.

Anywho, till then, I very much appreciate this project. Not that I don't wish I understood how to keep wifi from disabling, but I think that isn't your fault. Probably something I need to figure out that is more central to my own OS/hardware.

Have a good one!