AdnanHodzic/auto-cpufreq #397: auto-cpufreq won't install on Ubuntu JammyJellyFish

JPengu16, 2022-05-22:
### Error output:
```
----------------------------------- Warning -----------------------------------

Detected running GNOME Power Profiles daemon service!
This daemon might interfere with auto-cpufreq and should be disabled.

Steps to perform this action using auto-cpufreq: power_helper script:
git clone https://github.com/AdnanHodzic/auto-cpufreq.git
cd auto-cpufreq/auto_cpufreq
python3 power_helper.py --gnome_power_disable

----------------------------------- Warning -----------------------------------

Detected you are running a TLP service!
This daemon might interfere with auto-cpufreq which can lead to unexpected results.
We strongly encourage you to remove TLP unless you really know what you are doing.

-------------------------------------------------------------------------------

```
---

### System information:
CPU max frequency: 2800 MHz
CPU min frequency: 800 MHz

Core	Usage	Temperature	Frequency
CPU0:	  1.0%     39 °C     1800 MHz
CPU1:	  2.0%     39 °C     1713 MHz

auto-cpufreq version: 1.9.3 (git: ae1f53c7)

Python: 3.10.4
psutil package: 5.9.1
platform package: 1.0.8
click package: 8.1.3
distro package: 1.7.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.65 %

Total CPU usage: 2.5 %
Total system load: 0.9873046875
Average temp. of all cores: 39.0 °C 

Currently using: schedutil governor
Currently turbo boost is: on

#### Without providing requested information your issue might be ignored!

---

If installation failed with [auto-cpufreq-installer ](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer)have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?

prazwalayyyyyyyy, 2022-06-09:
I am currently having the same problem. just before a week the minimum clock speed my cpus ran was 600mHz and currently its on 400.

AdnanHodzic, 2022-06-09:
Please take a look at: [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq) & [Troubleshooting](https://github.com/AdnanHodzic/auto-cpufreq/#troubleshooting)

Also, include information that was requested from you when submitting an issue, i.e:

Add/paste output of:

```
auto-cpufreq --debug
auto-cpufreq --stats
```

AdnanHodzic, 2022-07-16:
Closing the issue due to inactivity and issue not answering issue template questions, reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

closed 2022-07-16 by AdnanHodzic