AdnanHodzic/auto-cpufreq #404: Turbo boost constantly toggles ON/OFF for <10% CPU load

DhruvaRajwade, 2022-06-22:
`Linux distro: EndeavourOS rolling rolling`
`Linux kernel: 5.18.5-arch1-1`
`Processor: AMD Ryzen 7 4800H with Radeon Graphics`
`Cores: 16`
`Architecture: x86_64`
`Driver: acpi-cpufreq`

Running auto-cpufreq --stats outputs the following:(Relevent part)

`Battery is: charging`

`Setting to use: "performance" governor`

`Total CPU usage: 11.7 %`
`Total system load: 2.34`
`Average temp. of all cores: 57.88 °C `

`High CPU load`
`setting turbo boost: on`

For the next refresh, the output:

`Battery is: charging`

`Setting to use: "performance" governor`

`Total CPU usage: 11.8 %`
`Total system load: 2.34`
`Average temp. of all cores: 75.62 °C `

High CPU load`
`Optimal total CPU usage: 11.8 %, high average core temp: 75.625 °C`
`setting turbo boost: off`

This is a pattern I have noticed where it constantly toggles Turbo boost off/On without much change in CPU usage. My concern is is this intentional/A bug and should I disable the auto-turbo boost i.e will the constant toggling potentially cause damage to my CPU?

AdnanHodzic, 2022-07-16:
Seems to be related to: #398

closed 2022-07-16 by DhruvaRajwade

zoitrok, 2022-07-17:
Looks like there's a [hardcoded threshold](https://github.com/AdnanHodzic/auto-cpufreq/blob/cde4cbeb45f12c168b2e58ecf20cf593f77c40a2/auto_cpufreq/core.py#L632) of 70⁰C to disable turbo boost.

You're hitting that threshold, so seems to be working as designed. I guess most computers would have some safety to clock down, albeit at a bit higher threshold. Maybe this threshold would be a good candidate to add to the configuration file?