AdnanHodzic/auto-cpufreq #424: wrong core frequnacy find for core i9 12900h

Maziar123, 2022-08-24:
Hi

I use last version in Manjaro
with kernel 5.19

conf:
governor = performance
scaling_min_freq = 2800000
scaling_max_freq = 4000000
seems max freq find wrong ...

Given value for 'scaling_max_freq' is not within the allowed frequencies, 400000-2901000 kHz

AdnanHodzic, 2022-08-24:
Template issue ignored and not filled out, hence making this issue invalid, closing.

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

closed 2022-08-24 by AdnanHodzic

Maziar123, 2022-08-25:
Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: Manjaro Linux 21.3.7 Ruah
Linux kernel: 5.19.1-3-MANJARO
Processor: 12th Gen Intel(R) Core(TM) i9-12900H
Cores: 20
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2901 MHz
CPU min frequency: 400 MHz

Core    Usage   Temperature     Frequency
CPU0:     1.0%     47 °C     2901 MHz
CPU1:     0.0%     47 °C     2901 MHz
CPU2:     0.0%     50 °C     2901 MHz
CPU3:     0.0%     50 °C     3483 MHz
CPU4:     1.0%     49 °C     3082 MHz
CPU5:     0.0%     49 °C     2901 MHz
CPU6:     0.0%     47 °C     2901 MHz
CPU7:     1.0%     47 °C     2901 MHz
CPU8:     0.0%     47 °C     2901 MHz
CPU9:     0.0%     47 °C     2901 MHz
CPU10:    0.0%     48 °C     2901 MHz
CPU11:    0.0%     48 °C     2901 MHz
CPU12:    0.0%     50 °C     2901 MHz
CPU13:    0.0%     50 °C     2901 MHz
CPU14:    0.0%     50 °C     2901 MHz
CPU15:    1.0%     50 °C     2901 MHz
CPU16:    0.0%     48 °C     2901 MHz
CPU17:    0.0%     48 °C     2901 MHz
CPU18:    0.0%     48 °C     2901 MHz
CPU19:    0.0%     48 °C     2901 MHz

auto-cpufreq version: Version         : 1.9.5-1

Python: 3.10.5
psutil package: 5.9.1
platform package: 1.0.8
click package: 8.1.3
distro package: 1.7.0

Computer type: Detachable
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.16 %

Total CPU usage: 0.2 %
Total system load: 0.38
Average temp. of all cores: 48.40 °C 

Currently using: performance governor
Currently turbo boost is: on

-------------------------------------------------------------------------------