AdnanHodzic/auto-cpufreq #427: Could it offer the sleep mode for a period of inactive operation ?

applejwjcat, 2022-08-31:
As above, if the laptop is inactive for 5 minutes, is there any operation that could let the machine move into suspend mode actomaticlly, like tlp do it similarly?

AdnanHodzic, 2022-08-31:
It does do this. Idea is for auto-cpufreq not to interfere into Linux/kernel doing the heavy lifting. auto-cpufreq will set the right governor/frequency during the idle time, cores will go to their minimum and device will go to suspend if that's what it's configured to do after 5 minutes.

With this said, will also close this issue as it's not an issue but rather a question which should've been posted in [Discussion](https://github.com/AdnanHodzic/auto-cpufreq/discussions) section.

applejwjcat, 2022-08-31:
> It does do this. Idea is for auto-cpufreq not to interfere into Linux/kernel doing the heavy lifting. auto-cpufreq will set the right governor/frequency during the idle time, cores will go to their [...]

Excuse me, But how to configure it to make the screen light down and the device suspended after a specific time? how to change that time, I mean in the conf file? 
Is there a complete configuration file or some man doc to understand ?