AdnanHodzic/auto-cpufreq #428: Systemd does not start service on Arch

stevensko, 2022-09-03:
actually endeavouros. following the instructions to install from the aur and then starting and enabling the systemd service for the daemon instead of the --install command. upon reboot, it says it is enabled but it is inactive (dead) with no failure codes. which makes me think it's not being triggered but the multiusertarget is active. it starts with systemctl start command but it does not stay started with a reboot.

stevensko, 2022-09-03:
I ran monitor and it says gnome power profiles is detected even though I'm on kde. going to investigate if that's the issue.

stevensko, 2022-09-03:
I left the gnome power profiles because other packages were dependent on it and I moved on

AdnanHodzic, 2022-09-03:
Template issue ignored and not filled out, hence making this issue invalid, closing.

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

closed 2022-09-03 by AdnanHodzic