AdnanHodzic/auto-cpufreq #429: Build broken on Arch

DaniAndTheWeb, 2022-09-06:
I'm not sure if there's been a breaking change in python but when built on arch linux the program doesn't seem to run as it doesn't display any stats using the --stats flag and throws this error when trying to check with the --monitor flag:

Traceback (most recent call last):
  File "/usr/bin/auto-cpufreq", line 226, in <module>
    main()
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/bin/auto-cpufreq", line 100, in main
    sysinfo()
  File "/usr/lib/python3.10/site-packages/auto_cpufreq/core.py", line 1104, in sysinfo
    cpu = int(coreid_info[i].split(":")[-1])
ValueError: invalid literal for int() with base 10: ' egrep is obsolescent; using grep -E'

AdnanHodzic, 2022-09-07:
Template issue ignored and not filled out, hence making this issue invalid, closing.

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

closed 2022-09-07 by AdnanHodzic

DaniAndTheWeb, 2022-09-07:
The output i posted is also the output that i obtain using the debug flag

DaniAndTheWeb, 2022-09-07:
The build process doesn't show any error, it only throws a couple of warnings about setup.py install and setuptools.installer being deprecated. Python version is 3.10.7