AdnanHodzic/auto-cpufreq #445: Not compatible with Fedora 37 - error with wheel package in install script

Gogny, 2022-10-26:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? -> YES
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? -> YES, got an error error: `system does not fully support snapd: cannot mount squashfs using "squashfs": `
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? -> no such file as it doesn't install
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? -> YES

### Error output:
It seems a package is depreciated this might be a repo problem, here is the output when I try to run the install script. It worked perfectly on Fedora 36, but since I don't have auto-cpufreq my laptop's battery life is well worse... 
```text
sudo ./auto-cpufreq-installer
i

Installing necessary Python packages

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib64/python3.11/site-packages (22.3)
Requirement already satisfied: setuptools in /opt/auto-cpufreq/venv/lib64/python3.11/site-packages (from -r requirements.txt (line 1)) (62.6.0)
Requirement already satisfied: psutil in /opt/auto-cpufreq/venv/lib64/python3.11/site-packages (from -r requirements.txt (line 2)) (5.9.2)
Requirement already satisfied: click in /opt/auto-cpufreq/venv/lib64/python3.11/site-packages (from -r requirements.txt (line 3)) (8.1.3)
Requirement already satisfied: distro in /opt/auto-cpufreq/venv/lib64/python3.11/site-packages (from -r requirements.txt (line 4)) (1.8.0)

-------------------------------------------------------------------------------

installing auto-cpufreq tool

/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      /opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/config/setupcfg.py:463: SetuptoolsDeprecationWarning: The license_file parameter is deprecated, use license_files instead.
        warnings.warn(msg, warning_class)
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help
      
      error: invalid command 'bdist_wheel'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for wrapt
ERROR: Failed to build one or more wheels
Traceback (most recent call last):
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/installer.py", line 82, in fetch_build_egg
    subprocess.check_call(cmd)
  File "/usr/lib64/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/opt/auto-cpufreq/venv/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpf6b4h10o', '--quiet', 'wrapt<2,>=1.10']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/romain/Téléchargements/auto-cpufreq/auto-cpufreq/setup.py", line 18, in <module>
    setup(
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/__init__.py", line 86, in setup
    _install_setup_requires(attrs)
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/__init__.py", line 80, in _install_setup_requires
    dist.fetch_build_eggs(dist.setup_requires)
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/dist.py", line 876, in fetch_build_eggs
    resolved_dists = pkg_resources.working_set.resolve(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/pkg_resources/__init__.py", line 789, in resolve
    dist = best[req.key] = env.best_match(
                           ^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/pkg_resources/__init__.py", line 1075, in best_match
    return self.obtain(req, installer)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/pkg_resources/__init__.py", line 1087, in obtain
    return installer(requirement)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/dist.py", line 956, in fetch_build_egg
    return fetch_build_egg(self, req)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/auto-cpufreq/venv/lib64/python3.11/site-packages/setuptools/installer.py", line 84, in fetch_build_egg
    raise DistutilsError(str(e)) from e
distutils.errors.DistutilsError: Command '['/opt/auto-cpufreq/venv/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpf6b4h10o', '--quiet', 'wrapt<2,>=1.10']' returned non-zero exit status 1.
```
When I try installing the package manually :
```
pip install wheel
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: wheel in /home/romain/.local/lib/python3.11/site-packages (0.37.1)
```
---

### System information:

```
auto-cpufreq --debug
/opt/auto-cpufreq/venv/bin/python: can't open file '/opt/auto-cpufreq/venv/bin/auto-cpufreq': [Errno 2] No such file or directory
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

SunkenHero, 2022-10-27:
mayby try install the wheel package with sudo to install it system wide `sudo pip install wheel`

Gogny, 2022-10-27:
Thank you @SunkenHero , sadly it seems it was already installed globally :  
``sudo pip3 install wheel  
Requirement already satisfied: wheel in /usr/lib/python3.11/site-packages (0.37.1)``
I even tried with `--force-reinstall` and `--ignore-installed`.
What is the wheel package used for in auto-cpufreq ? I went on [this page](https://pypi.org/project/wheel/) and it is written this :  

> It should be noted that wheel is not intended to be used as a library, and as such there is no stable, public API.
Anyone has a clue ?

SunkenHero, 2022-10-27:
Mayby try install the latest wheel version of fedora 36, if its possible with pip.

Gogny, 2022-10-27:
[image]
The last version was 1 year ago, so the latest version of Fedora 36 will be the same as the one I am using.
I think there is a problem with the installation script. Did you contribute on it ?

SunkenHero, 2022-10-28:
I didn't contributed to the installer script. In this region i don't know much and cant really help...

pakin1, 2022-10-28:
Same issue with Fedora 37. For now I'm using snap version.

SunkenHero, 2022-10-28:
Looks like on Fedora 37 the installation with the installer script doesn't work. pakin1 got it to work with snap, so mayby we should try to installed it with snap on your device

Gogny, 2022-10-30:
@pakin1 didn't you ever have an error like that : `system does not fully support snapd: cannot mount squashfs using "squashfs": ` ? It prevents me from installing any snap package...

SunkenHero, 2022-10-30:
[Installing snap on fedora](https://snapcraft.io/docs/installing-snap-on-fedora), look i the troubleshooting, i think thats pretty near to your problem...

pakin1, 2022-11-01:
> @pakin1 didn't you ever have an error like that : `system does not fully support snapd: cannot mount squashfs using "squashfs": ` ? It prevents me from installing any snap package...

Never had this kind of issue.

MrinmoyHaloi, 2022-11-06:
Any updates on it ?

SunkenHero, 2022-11-06:
> Any updates on it ?

It looks like the installer script doesn't work on fedora 37, but you can use the snap package.

MrinmoyHaloi, 2022-11-06:
I know but i don't use snap. A lot of people doesn't wants to use it either for various reasons. Ig the script can be fixed with some changes.

benezen, 2022-11-08:
now void has same issue

benezen, 2022-11-08:
fixed
removed python3-wheel python3-wrapt packages
and type in terminal
$ sudo pip3 install wheel wrapt --ignore-installed
and then installer works

MiMillieuh, 2022-11-08:
> 

That doesn't work for me...

MiMillieuh, 2022-11-08:
> > 
> 
> That doesn't work for me...

Using this branch worked for me, I guess there is something with wheel on Fedora 37 

https://github.com/johnseekins/auto-cpufreq

AdnanHodzic, 2022-11-11:
As a sidenote, if anyone here wants to give it a try in implementing this functionality as part of [auto-cpufreq-installer ](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto-cpufreq-installer)or [snap package,](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/snap/snapcraft.yaml) please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

SunkenHero, 2022-11-12:
> Using this branch worked for me, I guess there is something with wheel on Fedora 37
> 
> https://github.com/johnseekins/auto-cpufreq

johnseekins just changed the install file a little bit, by adding wheel to the pip install:
`python3 -m pip install --upgrade pip wheel`
This could be a global fix maybe we could implement it too.

AdnanHodzic, 2022-11-12:
> johnseekins just changed the install file a little bit, by adding wheel to the pip install: `python3 -m pip install --upgrade pip wheel` This could be a global fix maybe we could implement it too.

Yup, then this problem has been fixed and merged with auto-cpufreq `master` branch as part of #477

Closing the issue.

closed 2022-11-12 by AdnanHodzic