AdnanHodzic/auto-cpufreq #446: Warning: Changing CPU turbo is not supported. Skipping.

vinograd-cyber, 2022-10-27:
The installation was successful, but unfortunately a warning appeared, "**Warning: Changing CPU turbo is not supported. Skipping.**" Is it because my computer's cpu model is not supported?
```
root@debian:~# auto-cpufreq --version

-------------------------------------------------------------------------------

Linux distro: Debian GNU/Linux 11 bullseye
Linux kernel: 5.10.0-19-amd64
auto-cpufreq version: 1.9.6 (git: ff72e1e0)
```

-------------------------------------------------------------------------------
```
-------------------------------------------------------------------------------

                "auto-cpufreq" is about to refresh ...
                Executed on: October 27 (Thursday) - 10:21:39

-------------------------------------------------------------------------------

Linux distro: Debian GNU/Linux 11 bullseye
Linux kernel: 5.10.0-19-amd64
Processor: Intel(R) Core(TM) i3-4170 CPU @ 3.70GHz
Cores: 4
Architecture: x86_64
Driver: intel_cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3700 MHz
CPU min frequency: 800 MHz

Core    Usage   Temperature     Frequency
CPU0:     0.0%     29 °C     3692 MHz
CPU1:     1.0%     30 °C     3693 MHz
CPU2:     0.0%     29 °C     3693 MHz
CPU3:     1.0%     30 °C     3703 MHz

---------------------------- CPU frequency scaling ----------------------------

Setting to use: "performance" governor

Total CPU usage: 0.5 %
Total system load: 0.13
Average temp. of all cores: 29.50 °C 

Load optimal
setting turbo boost: off
Warning: Changing CPU turbo is not supported. Skipping.
```

cpuinfo：
```
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 60
model name      : Intel(R) Core(TM) i3-4170 CPU @ 3.70GHz
stepping        : 3
microcode       : 0x12
cpu MHz         : 3692.112
cache size      : 3072 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr s
se sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_
tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2
 movbe popcnt aes xsave avx f16c rdrand lahf_lm abm cpuid_fault invpcid_single pti tpr_shadow vnmi flexpriority ept v
pid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm arat pln pts
vmx flags       : vnmi preemption_timer invvpid ept_x_only ept_ad ept_1gb flexpriority tsc_offset vtpr mtf vapic ept 
vpid unrestricted_guest ple
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit srbds mmio_unkno
wn
bogomips        : 7383.32
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
--More--(-124400%)
```

vinograd-cyber, 2022-10-27:
root@debian:~# auto-cpufreq --debug

```
-------------------------------------------------------------------------------

Linux distro: Debian GNU/Linux 11 bullseye
Linux kernel: 5.10.0-19-amd64
Processor: Intel(R) Core(TM) i3-4170 CPU @ 3.70GHz
Cores: 4
Architecture: x86_64
Driver: intel_cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3700 MHz
CPU min frequency: 800 MHz

Core    Usage   Temperature     Frequency
CPU0:     0.0%     30 °C     3692 MHz
CPU1:     1.0%     30 °C     3689 MHz
CPU2:     1.0%     30 °C     3692 MHz
CPU3:     1.0%     30 °C     3692 MHz

auto-cpufreq version: 1.9.6 (git: ff72e1e0)

Python: 3.9.2
psutil package: 5.9.3
platform package: 1.0.8
click package: 8.1.3
distro package: 1.8.0

Computer type: Desktop
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.28 %

Total CPU usage: 0.5 %
Total system load: 0.00
Average temp. of all cores: 30.00 °C 

Currently using: performance governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```

vinograd-cyber, 2022-10-27:
Intel officially stated that this cpu does not support turbo boost, so the logic of your application is correct. Then, the question I am concerned about now is whether the automatic adjustment of the cpu can still be effectively performed in this case. Woolen cloth?
```
Advanced Technologies
Intel® Turbo Boost Technology ‡
No
Intel® Hyper-Threading Technology ‡
Yes
Intel® Transactional Synchronization Extensions
No
Intel® 64 ‡
Yes
Instruction Set
64-bit
Instruction Set Extensions
Intel® SSE4.1, Intel® SSE4.2, Intel® AVX2
Idle States
Yes
Enhanced Intel SpeedStep® Technology
Yes
Thermal Monitoring Technologies
Yes 
```
https://www.intel.sg/content/www/xa/en/products/sku/81209/intel-core-i34170t-processor-3m-cache-3-20-ghz/specifications.html?countrylabel=Asia%20Pacific
[image]

vinograd-cyber, 2022-10-28:
> Looks like its all working. Your CPU is scaling up to 3.70GHz, because you are in performance mode. If you set it in the config file right, your CPU should scale down to save battery life, if you run on battery.

Oh great, this software is amazing, I will keep following it. thanks for your help.

AdnanHodzic, 2022-11-11:
Closing the issue since problem seems to be resolved.

closed 2022-11-11 by AdnanHodzic