AdnanHodzic/auto-cpufreq #448: Installation issues with snap on kali

HediMuhamad, 2022-10-30:
# Overview
After installing _auto-cpufreq_ using _snap_ on Kali, I found that
1. I can't use it as normal command-line tool, it needs sudo to every options, and when I wanna give it sudo privilege, 'sudo: auto-cpufreq: command not found' will be printed,
2. I can't find it in services list.

### Command-line issue
Without _sudo_
```
└─$ auto-cpufreq --monitor                                                                                                                                                                                                                1 ⨯

--------------------------------- Root check ----------------------------------

ERROR:

Must be run root for this functionality to work, i.e: 
sudo auto-cpufreq

-------------------------------------------------------------------------------
```
With _sudo_
```
└─$ sudo auto-cpufreq --monitor                                                                                                                                                                                                         130 ⨯
sudo: auto-cpufreq: command not found
```
With _snap run_ and without _sudo_
```
└─$ snap run auto-cpufreq --monitor                                                                                                                                                                                                     130 ⨯

--------------------------------- Root check ----------------------------------

ERROR:

Must be run root for this functionality to work, i.e: 
sudo auto-cpufreq

-------------------------------------------------------------------------------
```

With _snap run_ and _sudo_
```
└─$ sudo snap run auto-cpufreq --monitor

Note: You can quit monitor mode by pressing "ctrl+c"

----------------------------------- Warning -----------------------------------

Unable to detect state of GNOME Power Profiles daemon service!
This daemon might interfere with auto-cpufreq and should be disabled.

Steps to perform this action using auto-cpufreq: power_helper script:
git clone https://github.com/AdnanHodzic/auto-cpufreq.git
cd auto-cpufreq/auto_cpufreq
python3 power_helper.py --gnome_power_disable

Reference: https://github.com/AdnanHodzic/auto-cpufreq#configuring-auto-cpufreq

----------------------------------- Warning -----------------------------------

Unable to detect if you are using a TLP service!
This daemon might interfere with auto-cpufreq which can lead to unexpected results.
We strongly encourage you not to use TLP unless you really know what you are doing.

-------------------------------------------------------------------------------

Linux distro: Kali GNU/Linux 2022.1
Linux kernel: 5.16.0-kali3-amd64
Processor: AMD Ryzen 5 4600H with Radeon Graphics
Cores: 12
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3000 MHz
CPU min frequency: 1400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  2.0%     35 °C     1397 MHz
CPU1:	  2.0%     35 °C     1400 MHz
CPU2:	  1.0%     35 °C     1400 MHz
CPU3:	  3.0%     35 °C     1400 MHz
CPU4:	  4.0%     35 °C     1400 MHz
CPU5:	  2.0%     35 °C     1400 MHz
CPU6:	  1.0%     35 °C     3000 MHz
CPU7:	  2.0%     35 °C     3000 MHz
CPU8:	  1.0%     35 °C     1400 MHz
CPU9:	  1.0%     35 °C     1400 MHz
CPU10:	  2.0%     35 °C     1400 MHz
CPU11:	  0.0%     35 °C     1400 MHz

---------------------------- CPU frequency scaling ----------------------------

Battery is: charging (84%)

Currently using: schedutil governor
Suggesting use of "performance" governor

Total CPU usage: 2.2 %
Total system load: 0.38
Average temp. of all cores: 34.85 °C 

Load optimal
suggesting to set turbo boost: on
Currently turbo boost is: on

-------------------------------------------------------------------------------
```

## Service issue
```
└─$ systemctl status auto-cpufreq       
Unit auto-cpufreq.service could not be found.
```

---

# Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted?  **Submitted but not answered**. 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer), have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? **Both were tested, with the repo installation I had some dependencies issue like in [#444](https://github.com/AdnanHodzic/auto-cpufreq/issues/444), but using snap no error was traced, but I faced the current issue.**
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? **output: _Unit power-profiles-daemon.service could not be found._**
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? **No, the problem not listed here**

### Error output:
```text
└─$ sudo auto-cpufreq --monitor                                                                                                                                                                                                           1 ⨯
sudo: auto-cpufreq: command not found
```
---

### System information:

└─$ lscpu                                                                                                                                                                                                                              
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         48 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  12
  On-line CPU(s) list:   0-11
Vendor ID:               AuthenticAMD
  Model name:            AMD Ryzen 5 4600H with Radeon Graphics
    CPU family:          23
    Model:               96
    Thread(s) per core:  2
    Core(s) per socket:  6
    Socket(s):           1
    Stepping:            1
    Frequency boost:     disabled
    CPU max MHz:         3000.0000
    CPU min MHz:         1400.0000
    BogoMIPS:            5988.73
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf
                          rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr
                         _core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba ibrs ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves c
                         qm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local clzero irperf xsaveerptr rdpru wbnoinvd arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload
                          vgif v_spec_ctrl umip rdpid overflow_recov succor smca
Virtualization features: 
  Virtualization:        AMD-V
Caches (sum of all):     
  L1d:                   192 KiB (6 instances)
  L1i:                   192 KiB (6 instances)
  L2:                    3 MiB (6 instances)
  L3:                    8 MiB (2 instances)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-11
Vulnerabilities:         
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; Full AMD retpoline, IBPB conditional, IBRS_FW, STIBP conditional, RSB filling
  Srbds:                 Not affected
  Tsx async abort:       Not affected

```
└─$ auto-cpufreq --debug

--------------------------------- Root check ----------------------------------

ERROR:

Must be run root for this functionality to work, i.e: 
sudo auto-cpufreq

-------------------------------------------------------------------------------

```

Then with _sudo_
```

└─$ sudo auto-cpufreq --debug                                                                                                                                                                                                             1 ⨯
sudo: auto-cpufreq: command not found

```

SunkenHero, 2022-10-31:
The Snap Package is very unstable and often don't work. You can try install it with the installer script.

AdnanHodzic, 2022-11-11:
> The Snap Package is very unstable and often don't work. You can try install it with the installer script.

@SunkenHero  This is not true, I spend a lot of time on making Snap package work, and even I like to use it to make sure everything is working as expected.

@HediMuhamad  as you stated, in case snap package fails just fallback to using [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer). 

Closing the issue due to inactivity.

closed 2022-11-11 by AdnanHodzic