AdnanHodzic/auto-cpufreq #451: How can I stop disabling of bluetooth on boot?

hardikprakash, 2022-11-07:
Thank you in advance!

AdnanHodzic, 2022-11-11:
This should be in [Discussion](https://github.com/AdnanHodzic/auto-cpufreq/discussions) instead. 

Regardless,  to answer your question:

> can I stop disabling of bluetooth on boot?

Quick answer is no. 

I made a [suggestion how this issue](https://github.com/AdnanHodzic/auto-cpufreq/issues/92#issuecomment-671425807) could be resolved in #92. I'm also open to an idea that bluetooth remains enabled on boot by default, but in this case to also have a parameter to disable it (as it is now).

Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

Mark3xtrm, 2023-11-11:
Hi! I edited main.conf setting back `AutoEnable=true`, but it seems bluetooth still remains off at boot. Any help?

Mark3xtrm, 2023-11-11:
> Hi! I edited main.conf setting back `AutoEnable=true`, but it seems bluetooth still remains off at boot. Any help?

Now it seems to work. I rebooted again. Please, add a toggle or an installation option for this. Like
`let bluetooth continue to start automatically at startup`

AdnanHodzic, 2023-11-12:
Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

Mark3xtrm, 2023-11-12:
> Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the [...]

Thanks, @AdnanHodzic , I'll see what I can do as soon as I have some time for me! 🔥

Just another curiosity: does auto-cpufreq interfere with other system default settings, like eg brightness steps control? Maybe I'm wrong or it's just an impression, but after I enabled it I wasn't able to fine adjust brightness like before due to reduced number of steps in the slider

AdnanHodzic, 2023-11-14:
> Just another curiosity: does auto-cpufreq interfere with other system default settings, like eg brightness steps control? 

No absolutely not, other then disabling bluetooth it won't touch anything else in "user space". 

> Maybe I'm wrong or it's just an impression, but after I enabled it I wasn't able to fine adjust brightness like before due to reduced number of steps in the slider

[As mentioned in Readme, auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#1-power_helperpy-script-snap-package-install-only) will [disable GNOME power profiles](https://twitter.com/fooctrl/status/1467469508373884933) to prevent conflicts and various other issues.

AdnanHodzic, 2025-04-13:
Merged _"Implement dedicated functionality to turn Bluetooth on/off at boot"_ #836 PR which implements requested functionality.

Please pull the latest changes.

closed 2025-04-13 by AdnanHodzic