AdnanHodzic/auto-cpufreq #467: [Feature] Performance Powersave toggler.

niksingh710, 2023-01-03:
The tools are working fine.
Just have one request it should have a cli way of interaction to switch the governor.

e.g 
currently, my config states that on charging the governor should be `performance` and when not on AC it should be `powersave`

This ramps up my fan while charging.
sometimes I want it to be on powersave even when it is charging to keep my fan silent. and for that I have to change the config and reload auto-cpufreq

I want the feature to be something like 
```bash
auto-cpufreq --force governer powersave    # This should keep the governer to powersave mode no matter what the state is.
auto-cpufreq --force governer performance # This should keep the governer to performance mode no matter what the state is (AC or on Battery)
auto-cpufreq --force reset # resets to the config state.
auto-cpufreq --state # (should return powersave,performance,reseted) So that the user can create a indicator in it's bar config.
```

AdnanHodzic, 2023-01-03:
This is already possible with [auto-cpufreq config file](https://github.com/AdnanHodzic/auto-cpufreq#2-auto-cpufreq-config-file). 

Otherwise since I'm very limited with available time to do this myself. If you want to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

AdnanHodzic, 2023-02-01:
Changes made by @shadeyg56 as part of #476 are now merged with `master` branch, hence closing this issue.

closed 2023-02-01 by AdnanHodzic