AdnanHodzic/auto-cpufreq #475: Install script fails on Pop OS 22.04

AMoonRabbit, 2023-01-20:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
installing auto-cpufreq tool

/opt/auto-cpufreq/venv/lib/python3.10/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(

```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

Don't get a chance to use the auto-cpufreq command as it fails to install before that using the script.  The error appears to be cut off, and not complete which does not help. 

Output from neofecth

```
OS: Pop!_OS 22.04 LTS x86_64 
Host: Laptop (12th Gen Intel Core) A4 
Kernel: 6.0.12-76060006-generic 
Uptime: 8 hours, 27 mins 
Packages: 2093 (dpkg), 20 (flatpak) 
Shell: zsh 5.8.1 
Resolution: 2256x1504 
DE: GNOME 42.3.1 
WM: Mutter 
WM Theme: Pop 
Theme: Orchis-Red-Dark-Compact [GTK2/3] 
Icons: candy-icons-master [GTK2/3] 
Terminal: tilix 
CPU: 12th Gen Intel i5-1240P (16) @ 1.700GHz 
GPU: Intel Alder Lake-P 
Memory: 5009MiB / 15708MiB 
```

AdnanHodzic, 2023-01-20:
Template issue ignored and not filled out, hence making this issue invalid, closing. Feel free to re-open this issue with properly filled out requested fields.

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

closed 2023-01-20 by AdnanHodzic

AMoonRabbit, 2023-01-21:
Excuse me, how am I supposed to fill the template, if CPUfreq does not install to the point I can run auto-cpufreq --debug? Please explain how I am supposed to meet this requirement?

AMoonRabbit, 2023-01-21:
I've managed to figure it out anyway. Something to do with my VPN and autocpu-freq not being able to get the dependencies.