AdnanHodzic/auto-cpufreq #477: [AUR] Maintained?

niksingh710, 2023-01-28:
is the AUR version of the tool is maintained?

AdnanHodzic, 2023-01-28:
Yes, it was already mentioned [on README](https://github.com/AdnanHodzic/auto-cpufreq/#aur-package-archmanjaro-linux). But I just pushed another change to make it even more explicit that it's not maintained.

Closing the issue ...

niksingh710, 2023-02-21:
@AdnanHodzic I was planning to maintain the Aur then I noticed that `auto-cpufreq-git` is working as expected and fine.
There are a few things the user has to keep in mind if installing from aur.

If facing `psutil` error: 
> Make sure pip version of psutil is not installed in system. Either install it via aur `python-psutil` or let the `auto-cpufreq-git` install it.

The user does not require to use the command `auto-cpufreq install` as the aur version installs it already.
The user just need to enable the system service. `sudo systemctl enable --now auto-cpufreq`.

AdnanHodzic, 2023-02-21:
@niksingh710 

> I was planning to maintain the Aur 

Awesome! 

> I noticed that auto-cpufreq-git is working as expected and fine.

:+1: 

> If facing `psutil` error:
> 
> > Make sure pip version of psutil is not installed in system. Either install it via aur `python-psutil` or let the `auto-cpufreq-git` install it.
>  
Can't you just install `python-psutil` as part of dependencies for AUR?

> The user does not require to use the command `auto-cpufreq install` as the aur version installs it already. The user just need to enable the system service. `sudo systemctl enable --now auto-cpufreq`.

Yea, I remember that being part of discussion #84, which is fine then we just note that as part of Readme?

niksingh710, 2023-02-21:
> Can't you just install python-psutil as part of dependencies for AUR?

`python-psutil` is a part of dependencies in the AUR.
The installation throws error if user have already installed `psutils` from pip.

> then we just note that as part of Readme?

Yep, should add it in Readme.

AdnanHodzic, 2023-02-21:
> `python-psutil` is a part of dependencies in the AUR.
> The installation throws error if user have already installed `psutils` from pip.

Ah, not sure in AUR packaging, but if possible as it is for Debian i,e as part of `Pre-Depends` to do some kind of custom bash logic to check if pip version is installed and if it is to remove it. As auto-cpufreq-installer will install it to a Python virtual env and system wide, so it won't be coming from here.

AdnanHodzic, 2024-05-25:
Since AUR is nowadays fully maintained, if you have any questions please feel free to contact one of the AUR maintainers: https://github.com/AdnanHodzic/auto-cpufreq/?tab=readme-ov-file#aur-package-arch-based-distributions

closed 2024-05-25 by AdnanHodzic