AdnanHodzic/auto-cpufreq #479: Auto-cpufreq is running in daemon mode

Gautameshwar, 2023-01-30:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted?  yes. submitted once, but not solved
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? yes. both of them had this issue
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? yes
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? no (its not related to this issue)

### Problem: 
I have issues running auto-cpu freq --install because that is needed for viewing CPU stats. However, the error says its running in Daemon mode and cannot do it. Please help me out on how to disable daemon and run this! 

---
### Error output:
```text
(base) gauthameshwar@gauthameshwar-ideapad-Flex-5:~/auto-cpufreq/auto_cpufreq$ sudo auto-cpufreq --install

------------------------ auto-cpufreq running ------------------------------

ERROR: auto-cpufreq is running in daemon mode.

Make sure to stop the daemon before running with --live or --monitor mode

-------------------------------------------------------------------------------

```
---

### System information:

Add/paste output of:

```
(base) gauthameshwar@gauthameshwar-ideapad-Flex-5:~/auto-cpufreq/auto_cpufreq$ sudo auto-cpufreq --debug

-------------------------------------------------------------------------------

Linux distro: Ubuntu 22.10 Kinetic Kudu
Linux kernel: 5.19.0-29-generic
Processor: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2400 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  1.0%     42 °C      659 MHz
CPU1:	  2.1%     43 °C      751 MHz
CPU2:	  0.0%     41 °C     2400 MHz
CPU3:	  0.0%     42 °C     2130 MHz
CPU4:	  1.0%     42 °C     2400 MHz
CPU5:	  0.0%     43 °C     2400 MHz
CPU6:	  1.0%     41 °C     2400 MHz
CPU7:	  0.0%     42 °C     2400 MHz

auto-cpufreq version: 1.9.7 (git: f66df2a8)

Python: 3.10.7
psutil package: 5.9.4
platform package: 1.0.8
click package: 8.1.3
distro package: 1.8.0

Computer type: Convertible
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.3 %

Total CPU usage: 0.4 %
Total system load: 0.35
Average temp. of all cores: 42.00 °C 

Currently using: performance governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

AdnanHodzic, 2023-01-31:
If you don't want to run auto-cpufreq as a daemon then you need to run `sudo auto-cpufreq --remove` as [described on Readme](https://github.com/AdnanHodzic/auto-cpufreq/#install---auto-cpufreq-daemon). 

Closing the issue since it's not an issue but more of a question which should be posted on [Discussions page](https://github.com/AdnanHodzic/auto-cpufreq/discussions).

Gautameshwar, 2023-02-03:
Yes, as I have filled in the template, I have tried running that line in terminal but, it doesn't detect it! Look at the code below for reference:

```
(base) gauthameshwar@gauthameshwar-ideapad-Flex-5:/home$ sudo auto-cpufreq --remove
[sudo] password for gauthameshwar: 

auto-cpufreq daemon is not installed.

(base) gauthameshwar@gauthameshwar-ideapad-Flex-5:/home$ sudo auto-cpufreq --install

------------------------ auto-cpufreq running ------------------------------

ERROR: auto-cpufreq is running in daemon mode.

Make sure to stop the daemon before running with --live or --monitor mode
```

AdnanHodzic, 2023-02-03:
@Gautameshwar First make sure you have latest changes (as `f66df2a8` is 2 commits behind) as I wasn't able to reproduce this problem. @shadeyg56 are you able to reproduce this?

Then can you make sure that you don't have both packages installed, i.e: run `sudo snap remove auto-cpufreq`, and `sudo ./auto-cpufreq-installer --remove`

Both of these will remove any instances of auto-cpufreq running (including daemon). You can verify this with `ps -ef | grep auto-cpufreq` and if daemon indeed was running you'd see something like: `root      376501       1  1 13:39 ?        00:00:01 /opt/auto-cpufreq/venv/bin/python /opt/auto-cpufreq/venv/bin/auto-cpufreq --daemon`

Once you see that daemon is really not running install auto-cpufreq with i.e: ``sudo ./auto-cpufreq-installer --install`, followed by installing the daemon `sudo auto-cpufreq --install` and everything should work as it should.

shadeyg56, 2023-02-03:
I will test this later to see if I can reproduce

In the meantime, you can also try `systemctl status auto-cpufreq` (if using systemd), to see if the daemon is installed on your system, or if there has been an error involving the daemon. If you're not using systemd as your init system, then change the command accordingly

AdnanHodzic, 2023-02-03:
> In the meantime, you can also try `systemctl status auto-cpufreq` (if using systemd), to see if the daemon is installed on your system, or if there has been an error involving the daemon. If you're not using systemd as your init system, then change the command accordingly

Correct, all of this is [described as part of Readme](https://github.com/AdnanHodzic/auto-cpufreq/#install---auto-cpufreq-daemon) as I previously pointed out.

shadeyg56, 2023-02-03:
Not able to reproduce.

Please ensure you have pulled the latest changes and run `sudo ./auto-cpufreq-installer --install` to apply them

AdnanHodzic, 2023-02-05:
Closing due to inactivity, if problem persist please feel free to re-open the issue.

closed 2023-02-05 by AdnanHodzic