AdnanHodzic/auto-cpufreq #489: auto-cpu does not use my configuration ?

datthanhdoan, 2023-02-22:
### System information:

Add/paste output of:

```

Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: Ubuntu 22.04 Jammy Jellyfish
Linux kernel: 5.19.0-32-generic
Processor: AMD Ryzen 7 5800H with Radeon Graphics
Cores: 16
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3200 MHz
CPU min frequency: 3200 MHz

Core	Usage	Temperature	Frequency
CPU0:	 1.0%     33 °C     3200 MHz
CPU1:	 1.0%     33 °C     3200 MHz
CPU2:	 0.0%     33 °C     3200 MHz
CPU3:	 0.0%     33 °C     3200 MHz
CPU4:	 0.0%     33 °C     1947 MHz
CPU5:	 0.0%     33 °C     3200 MHz
CPU6:	 0.0%     33 °C     2085 MHz
CPU7:	 0.0%     33 °C     2427 MHz
CPU8:	 0.0%     33 °C     3200 MHz
CPU9:	 4.0%     33 °C     3200 MHz
CPU10:	 0.0%     33 °C     3200 MHz
CPU11:	 2.0%     33 °C     1397 MHz
CPU12:	 0.0%     33 °C     3200 MHz
CPU13:	 0.0%     33 °C     3200 MHz
CPU14:	 0.0%     33 °C     3200 MHz
CPU15:	 0.0%     33 °C     3200 MHz

auto-cpufreq version: 1.9.7 (git: 9912f9d8)

Python: 3.10.6
psutil package: 5.9.4
platform package: 1.0.8
click package: 8.1.3
distro package: 1.8.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.15 %

Total CPU usage: 0.6 %
Total system load: 0.69
Average temp. of all cores: 32.85 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```

There is my configtion :
```
# settings for when connected to a power source
[charger]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor.
governor = performance

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_min_freq = 800000

# maximum cpu frequency (in kHz)
# example: for 1GHz = 1000 MHz = 1000000 kHz -> scaling_max_freq = 1000000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_max_freq = 1000000

# turbo boost setting. possible values: always, auto, never
turbo = auto

# settings for when using battery power
[battery]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor
governor = powersave

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
scaling_min_freq = 1500000

# maximum cpu frequency (in kHz)
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# example: for 1GHz = 1000 MHz = 1000000 kHz -> scaling_max_freq = 1000000
# to use this feature, uncomment the following line and set the value accordingly
scaling_max_freq = 2000000

# turbo boost setting. possible values: always, auto, never
turbo = auto
```
The first time i run autocpu
```
Using settings defined in /etc/auto-cpufreq.conf file

Note: You can quit stats mode by pressing "ctrl+c"

-------------------------------------------------------------------------------

Linux distro: Ubuntu 22.04 Jammy Jellyfish
Linux kernel: 5.19.0-32-generic
Processor: AMD Ryzen 7 5800H with Radeon Graphics
Cores: 16
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3200 MHz
CPU min frequency: 3200 MHz

Core	Usage	Temperature	Frequency
CPU0:	 0.0%     31 °C     3200 MHz
CPU1:	 0.0%     31 °C     3200 MHz
CPU2:	 0.0%     31 °C     3200 MHz
CPU3:	 0.0%     31 °C     3200 MHz
CPU4:	 1.0%     31 °C     3200 MHz
CPU5:	 0.0%     31 °C     3200 MHz
CPU6:	 0.0%     31 °C     1397 MHz
CPU7:	 0.0%     31 °C     3200 MHz
CPU8:	 0.0%     31 °C     3200 MHz
CPU9:	 1.0%     31 °C     3200 MHz
CPU10:	 2.0%     31 °C     1547 MHz
CPU11:	 0.0%     31 °C     3200 MHz
CPU12:	 0.0%     31 °C     1397 MHz
CPU13:	 0.0%     31 °C     3200 MHz
CPU14:	 0.0%     31 °C     3200 MHz
CPU15:	 0.0%     31 °C     3200 MHz

---------------------------- CPU frequency scaling ----------------------------

Battery is: discharging

Setting to use: "powersave" governor
Setting maximum CPU frequency to 2000 Mhz
Setting minimum CPU frequency to 1500 Mhz

Total CPU usage: 0.9 %
Total system load: 0.52
Average temp. of all cores: 30.85 °C 

Load optimal (load average: 0.52, 0.44, 0.39)
setting turbo boost: off

-------------------------------------------------------------------------
```
The max and min allways is 3200MHz , completely different from my configuration
This continues when i change min and max cpu info

---

AdnanHodzic, 2023-03-11:
There is a bug when this feature was implemented, where to apply auto-cpufreq config file, you'll either need to stop/start auto-cpufreq process or restart after you make changes to the config file.

Can you confirm if changes are still not picked up you done what I suggested above?

Crusher50, 2023-03-13:
Doing what you suggested does apply the boosting config and governor, but the maximum scaling frequency still isn't respected after uninstalling and re-installing auto-cpufreq and a system restart.

mirocw, 2023-10-15:
I also have the same issue. It's not respecting my config file. By default mine cpu min is 1.2ghz (1200000 khz) and max is 3.3ghz (3300000 khz). When I tried to edit min to 1.6ghz, it does not follow instructions, instead min and max are at same level (3.3ghz)

Also for what ever reason when I'm on battery and do the stress cpu test, my cpu is staying at min cpu 1.2ghz
https://github.com/AdnanHodzic/auto-cpufreq/issues/583

datthanhdoan, 2023-10-15:
> There is a bug when this feature was implemented, where to apply changes made to auto-cpufreq config file, you'll either need to stop/start auto-cpufreq process or restart your computer after you make changes to the config file.
> 
> Can you confirm if changes are still not picked up you done what I suggested above?

Sorry for the late reply, I have tried restarting my laptop, deleting the config file and recreating it, modifying the config file, using only powersave or performance mode, and updating to a newer version (at this time it is 2.0) but it still didn't help.

expl0r3rgu1, 2023-12-18:
Try removing and reinstalling the daemon, after making changes to the config file, with:
`sudo auto-cpufreq --remove && sudo auto-cpufreq --install`

nghiapq77, 2024-05-08:
> There is a bug when this feature was implemented, where to apply changes made to auto-cpufreq config file, you'll either need to stop/start auto-cpufreq process or restart your computer after you [...]

Can confirm this works in my case.
`sudo systemctl restart auto-cpufreq`

AdnanHodzic, 2024-05-08:
Can you please make sure you have the latest changes pull from `master` branch? After merge of #663 changes made to config file should be automatically picked up without having to do anything ...

datthanhdoan, 2024-05-08:
I am no longer using Linux for my work, so I'm not sure if this issue has been fixed (but I believe it has been fixed based on the latest updates). I will close this issue so that the maintainer can focus on other issues

closed 2024-05-08 by datthanhdoan