AdnanHodzic/auto-cpufreq #496: Scaling number of cores?

123batman, 2023-03-27:
Does auto-cpufreq have an option to scale the number of cores?

Is it possible to support this feature?

What are the reasons not to have this feature if you have already though about it (as I am sure you have)?

My use case is focused on reducing energy consumption on workstations, not extending battery life on laptops.

AdnanHodzic, 2023-03-28:
This same request was made as part of #229, so I'll close this issue and let's continue discussion there.

> What are the reasons not to have this feature if you have already though about it (as I am sure you have)?

Answer is simple, not having enough time to develop one such future. Hence as I said in #229 if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

AdnanHodzic, 2024-03-10:
Until further notice #603 can be used as a reference how to manually disable CPU cores.