AdnanHodzic/auto-cpufreq #499: ResolutionError after installing auto-cpufreq using auto-cpufreq-installer

PhilippFeO, 2023-04-16:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
I have found #444 but I think it makes more sense to open a new issue to include my output neatly and using your template.
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package]
(https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
No, because I switched from Ubuntu to Linux Mint to avoid snap.
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
No, because auto-cpufreq doesn't start in the first place.
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?
Yes, but no proposals apply for my scenario.

### Error output:
I receive the following error message on every option. This one is from `auto-cpufreq --help`:
```text
Traceback (most recent call last):
  File "/opt/auto-cpufreq/venv/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.9.7+9912f9d8', 'auto-cpufreq')
  File "/opt/auto-cpufreq/venv/lib/python3.10/site-packages/pkg_resources/__init__.py", line 656, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/opt/auto-cpufreq/venv/lib/python3.10/site-packages/pkg_resources/__init__.py", line 1441, in run_script
    raise ResolutionError(
pkg_resources.ResolutionError: Script 'scripts/auto-cpufreq' not found in metadata at '/opt/auto-cpufreq/auto_cpufreq.egg-info'
```
---

### System information:

Add/paste output of: `auto-cpufreq --debug`

```
  $ auto-cpufreq --debug
Traceback (most recent call last):
  File "/opt/auto-cpufreq/venv/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.9.7+9912f9d8', 'auto-cpufreq')
  File "/opt/auto-cpufreq/venv/lib/python3.10/site-packages/pkg_resources/__init__.py", line 656, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/opt/auto-cpufreq/venv/lib/python3.10/site-packages/pkg_resources/__init__.py", line 1441, in run_script
    raise ResolutionError(
pkg_resources.ResolutionError: Script 'scripts/auto-cpufreq' not found in metadata at '/opt/auto-cpufreq/auto_cpufreq.egg-info'
```

Output of `uname -a`:
```text
Linux lenovo 5.15.0-69-generic #76-Ubuntu SMP Fri Mar 17 17:19:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

I tried removing and reinstalling it without success.

---

# Additional information

- I cloned the repository into `/opt`.
- There is an error message when removing auto-cpufreq, s. [Removing auto-cpufreq](#removing-auto-cpufreq)
- After removing auto-cpufreq, I can't uninstall the python packages (I copied `requirements.txt`):
```text
  $ pip uninstall -r requirements.txt 
Found existing installation: setuptools 59.6.0
Not uninstalling setuptools at /usr/lib/python3/dist-packages, outside environment /usr
Can't uninstall 'setuptools'. No files were found to uninstall.
Found existing installation: psutil 5.9.0
Not uninstalling psutil at /usr/lib/python3/dist-packages, outside environment /usr
Can't uninstall 'psutil'. No files were found to uninstall.
Found existing installation: click 8.0.3
Not uninstalling click at /usr/lib/python3/dist-packages, outside environment /usr
[... 180 lines cut ...]
Adding distro 1.8.0 to easy-install.pth file
Installing distro script to /opt/auto-cpufreq/venv/bin

Using /opt/auto-cpufreq/venv/lib/python3.10/site-packages
Searching for click==8.1.3
Best match: click 8.1.3
Adding click 8.1.3 to easy-install.pth file

Using /opt/auto-cpufreq/venv/lib/python3.10/site-packages
Finished processing dependencies for auto-cpufreq==1.9.7+9912f9d8
writing list of installed files to 'files.txt'

───────────────────────────────────────────────────────────────────────────────────────────────────────────

auto-cpufreq tool successfully installed.

For list of options, run:
auto-cpufreq --help"

───────────────────────────────────────────────────────────────────────────────────────────────────────────
```

---

It's your time you spent for a random guy on the internet, so thank you for reading my long issue!

rforced, 2023-04-29:
I'm having the same issue on PopOS 22.04 LTS

bar-julio, 2023-05-02:
I'm having the same issue on a fresh Ubuntu install

`uname -a`

`Linux HP-Laptop-17 6.2.0-20-generic #20-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr  6 07:48:48 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

Tried it with the snap package and the git clone install script

After rebooting and doing: 

`sudo systemctl status auto-cpufreq.service`

I get:

`Unit auto-cpufreq.service could not be found.`

AdnanHodzic, 2023-05-03:
@bar-julio 

> Tried it with the snap package and the git clone install script

As stated on [README](https://github.com/AdnanHodzic/auto-cpufreq), if install was performed as part of snap package daemon status can be verified by running: `systemctl status snap.auto-cpufreq.service.service`

AdnanHodzic, 2023-05-26:
Closing the issue due to inactivity.

PhilippFeO, 2023-05-26:
> Closing the issue due to inactivity.

My issue is still open. You responded to bar-julio (s. https://github.com/AdnanHodzic/auto-cpufreq/issues/499#issuecomment-1531945568), not my initial post.

fourteenminusone, 2023-06-20:
Same issue here on Debian Testing

SimpleCookie, 2023-12-27:
For the record, I have the same issue. I can't even seem to remove it completely. I want to uninstall so I can reinstall it, but the installation seems corrupt.

[image]

AdnanHodzic, 2023-12-27:
Could you please manually run [lines 42, 45, 48 & 51?](https://github.com/AdnanHodzic/auto-cpufreq/blob/8938cc92d74a97b42cdc079db04cb87c8c90dea3/scripts/auto-cpufreq-remove.sh#L42)

After this try re-running ` sudo ./auto-cpufreq-installer --remove` in location where auto-cpufreq is cloned. If same thing happens again please share so I see what happened. Then run `sudo rm -rf /opt/auto-cpufreq/` and reboot.

SimpleCookie, 2023-12-28:
> Could you please manually run [lines 42, 45, 48 & 51?](https://github.com/AdnanHodzic/auto-cpufreq/blob/8938cc92d74a97b42cdc079db04cb87c8c90dea3/scripts/auto-cpufreq-remove.sh#L42)
>
> After this try [...]

I think I might've resolved my issues somehow. Otherwise I'll do the steps and comment here again

SimpleCookie, 2023-12-28:
> Could you please manually run [lines 42, 45, 48 & 51?](https://github.com/AdnanHodzic/auto-cpufreq/blob/8938cc92d74a97b42cdc079db04cb87c8c90dea3/scripts/auto-cpufreq-remove.sh#L42)
>
> After this try [...]

Seems like it works better now.
Do I just run the install part now?

AdnanHodzic, 2023-12-29:
> Seems like it works better now. Do I just run the install part now?

Yep!

AdnanHodzic, 2024-05-25:
Closing due to inactivity.

closed 2024-05-25 by AdnanHodzic