AdnanHodzic/auto-cpufreq #518: make appimage or binary file, dont use snap , i use kali

newyorkthink, 2023-06-04:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---

AdnanHodzic, 2023-06-24:
There are [3 ways of installing auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#installing-auto-cpufreq), snap package, auto-cpufreq-installer and AUR package. Two of those options you can already use on Kali.

I don't intend to create an appimage, but if you or anyone else would like to create it and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

newyorkthink, 2023-06-24:
> There are [3 ways of installing auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#installing-auto-cpufreq), snap package, auto-cpufreq-installer and AUR package. Two of those options you can already use on Kali.有3种方法可以安装自动cpufreq，snap软件包，自动cpufreq安装程序和AUR软件包。其中两个选项你已经可以在Kali上使用。
> 
> I don't intend to create an appimage, but if you or anyone else would like to create it and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).我不打算创建应用程序图像，但是如果您或其他任何人想创建它并为该项目做出贡献，并且您将在未来版本中因您的工作而获得荣誉。

now i dont use auto-cpufreq, i use powertop,and linux-cpupower , i build it appimage ,is good

closed 2023-06-24 by AdnanHodzic