AdnanHodzic/auto-cpufreq #522: Make use of `energy_perf_bias`?

ret2src, 2023-06-13:
Hi,

first of all: Thank you for creating `auto-cpufreq`, I really like it and prefer it to more invasive alternatives like TLP.

I'm fairly new to Linux power management and just found out about the `energy_perf_bias` value for Intel processors (see <https://wiki.archlinux.org/title/CPU_frequency_scaling#Intel_performance_and_energy_bias_hint>).

Currently, it is set to the default value of `6` on my system although `auto-cpufreq` set the governor to `performance`.
I wonder if setting the value to `0` has the potential to further improve performance? If so, would it make sense to include such a feature in `auto-cpufreq`?

AdnanHodzic, 2023-06-24:
Could be an interesting thing to have ... since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

AdnanHodzic, 2024-09-09:
This functionality has been implemented as part of #771, please refer to [config file ](https://github.com/AdnanHodzic/auto-cpufreq?tab=readme-ov-file#3-auto-cpufreq-config-file)on how to enable & set it up.

closed 2024-09-09 by AdnanHodzic

ret2src, 2024-09-10:
Awesome, thank you!

PR #771 (2024-09-09): Add EPB (Energy Performance Bias) settings support
This PR adds support for the `energy_perf_bias` setting, giving users more control over how their Intel CPU (`intel_pstate` driver) balances performance and power efficiency

What's Changed:
- Introduced a new `energy_perf_bias` option, which can be set as a number (0-15) or using predefined strings like "performance", "balance_performance", "default", "balance_power" and "power" (made their names the same as EPP's)
- Integrated the `energy_perf_bias` setting into the existing power management logic, dynamically adjusting it based on the selec [... cut]