AdnanHodzic/auto-cpufreq #525: Why CPU Freq Crosses Hardware Limits and Policy Limits, and System Heat Up after Hybrid Sleep, in Debian 12 (Testing-Weekly) with auto-cpufreq?

aluva, 2023-06-19:
Installed `auto-cpufreq` and `thermald`, with `auto-cpufreq.conf `while on charger:

```
governor = performance
scaling_max_freq = 1900000
turbo = auto
```

And, `/etc/default/grub` edited as:

`GRUB_CMDLINE_LINUX_DEFAULT="quiet splash intel_pstate=disable"`

Thought it is conflicting/ redundant with some other tools. So, **removed** all other tools: ``tlp cpufreq cpupower acpi acpid acpi-support linux-cpupower cpufrequtils cpulimit`

OS: Debian 12 (bookworm: testing-weekly)
Hardware: (Apple: amd64, 16 GB RAM, 8 CPU) limit: 800 MHz - 2800 MHz

**After each reboot**, everything (CPU Freq, Temperature, etc.) stays within limits, as set in auto-cpufreq.conf.

But, **AFTER each Hybrid Sleep**, things go out of control: (CPU Freq near 3800 MHz, Temp 90℃, Fan 4600 RPM, etc).

What am I missing?

AdnanHodzic, 2023-06-24:
I never encountered anything like that during suspend, hibernate or hybrid sleep and settings stay as they are.

Could you fill out template when you were submitting an issue, this template is there for a reason. Thanks!

Reference:

### Fill out informtion requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
Add/paste error output in case of failed installation or other failing component
```
---

### System information:

Add/paste output of:

```
auto-cpufreq --debug
```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behavior.

aluva, 2023-06-25:
I resolved it.
I will write details about it. **Perhaps, it can be used as an 'aside'.**
Because, I noticed that many people struggle with similar issues..
**Adnan: With gratitude and immense thanks, for this path-breaking tool. Hats Off to you and the team working on it !!**

AdnanHodzic, 2023-06-30:
Most welcome! Feel free to [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution), either in code or updating the [Readme as part of troubleshooting section?](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

closed 2023-06-30 by AdnanHodzic