AdnanHodzic/auto-cpufreq #530: autocpufreq --version requires sudo now?

niksingh710, 2023-06-26:
any command with autocpu-freq requires sudo?

rootCircle, 2023-06-28:
Looks like a bug from my previous PR #524 . Searching through the code, I got to know that the `set_override` function is always called irrespective of the force flag is called or not. [Check this](https://github.com/AdnanHodzic/auto-cpufreq/blame/486a9a6dc21b742e1a85db681ac1368a5ef59912/bin/auto-cpufreq#L48).

A fix may be either :
1) To call `set_override` only when force has some value(non-null). Because, calling it for None values increases extra overhead with no added benefits associated with it. 
2) To prematurely return in `set_override` [function](https://github.com/AdnanHodzic/auto-cpufreq/blob/486a9a6dc21b742e1a85db681ac1368a5ef59912/auto_cpufreq/core.py#L98), if `override` value is None or illegal. 

I will try working on the issue and make an appropriate PR ASAP :-)

Till then you can just add just an indent on [this line](https://github.com/AdnanHodzic/auto-cpufreq/blob/486a9a6dc21b742e1a85db681ac1368a5ef59912/bin/auto-cpufreq#L48).

niksingh710, 2023-06-28:
> Looks like a bug from my previous PR #524 . Searching through the code, I got to know that the `set_override` function is always called irrespective of the force flag is called or not. [Check this](https://github.com/AdnanHodzic/auto-cpufreq/blame/486a9a6dc21b742e1a85db681ac1368a5ef59912/bin/auto-cpufreq#L48).
> 
> A fix may be either :
> 
>     1. To call `set_override` only when force has some value(non-null). Because, calling it for None values increases extra overhead with no added benefits associated with it.
> 
>     2. To prematurely return in `set_override` [function](https://github.com/AdnanHodzic/auto-cpufreq/blob/486a9a6dc21b742e1a85db681ac1368a5ef59912/auto_cpufreq/core.py#L98), if `override` value is None or illegal.
> 
> 
> I will try working on the issue and make an appropriate PR ASAP :-)
> 
> Till then you can just add an extra (tab) indent on [this line](https://github.com/AdnanHodzic/auto-cpufreq/blob/486a9a6dc21b742e1a85db681ac1368a5ef59912/bin/auto-cpufreq#L48). (calling `set_override` only if `force` is not None)

Premature return will be the preferred option imo as then it will depend on the `set_override` to handle the values and the core logic have to just call it instead of going over all the checks. Kinda will keep it clean.

More over a check for non-null will also work but not that much what I would like to see.

Btw if you do fix it and create a pr tag me as rn have added sudo's in all of my scripts will refactor them.

@AdnanHodzic opinion will matter more over it I guess.

rootCircle, 2023-06-28:
> Premature return will be the preferred option imo as then it will depend on the `set_override` to handle the values and the core logic have to just call it instead of going over all the checks. Kinda will keep it clean.

I guess it might be a better design, but at cost of an extra function call. But going through the codes and patterns, they preferred to include root_check in the main file, and not with the associated function. (eg: `remove_daemon` etc)

Due, to this I preferred going with 1st approach with a bit of modifications

> @AdnanHodzic opinion will matter more over it I guess.

It certainly would, I had created an PR and will see what he had to comment on this.

PR #531 (2023-06-30): [BUG-FIX] Fixed bug related to recurring sudo prompt for functions that even don't require it
The `set_override` function is always called irrespective of the force flag is called or not. This in turn, required root checks for every flag calls, even for `auto-cpufreq --version`.

To fix it, now `set_override` method is now called only if force flag is invoked.
Also root_check is moved from core.py to the main file, to ensure consistency with previous usage of root_check in the code used earlier.

Here are the specific code changes made:

In auto_cpufreq/core.py:
- Line 96: Removed the root_che [... cut]
commit c34e3e7f (2023-06-30): Fix bug related to repeated sudo prompt due to set_override calls (#531)

set_override method is now called only if force flag is invoked.

Here are the specific code changes made:

In auto_cpufreq/core.py:
- Line 96: Removed the root_check call from the set_override function.
           This was intended to be done to comply with code template,
           i.e., to use root_check in the main file.

In bin/auto-cpufreq:
- Line 45: Added a root_check call before calling set_override in main program.
           Also, set_override is only called if force option is invoked, saving
           us fro [... cut]
closed 2023-06-30 by AdnanHodzic