AdnanHodzic/auto-cpufreq #545: Temperatures reading issue

JosephBoom02, 2023-09-05:
### System information:

Linux distro: EndeavourOS rolling rolling
Linux kernel: 6.4.12-zen1-1-zen
Processor: AMD Ryzen 7 5800H with Radeon Graphics
Cores: 16
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3200 MHz
CPU min frequency: 1200 MHz

Core	Usage	Temperature	Frequency
CPU0      1.0%        43 °C      4374 MHz
CPU1      0.0%        43 °C      2566 MHz
CPU2     10.1%        43 °C      3901 MHz
CPU3      1.0%        43 °C      3513 MHz
CPU4      2.0%        43 °C      1200 MHz
CPU5      1.0%        43 °C      1790 MHz
CPU6      2.0%        43 °C      3513 MHz
CPU7      6.1%        43 °C      2848 MHz
CPU8      1.0%        43 °C      3503 MHz
CPU9      0.0%        43 °C      1197 MHz
CPU10      2.0%        43 °C      1200 MHz
CPU11      0.0%        43 °C      1200 MHz
CPU12      5.0%        43 °C      1198 MHz
CPU13      1.0%        43 °C      1200 MHz
CPU14      4.0%        43 °C      4374 MHz
CPU15      2.0%        43 °C      4374 MHz

CPU fan speed: 2331 RPM

auto-cpufreq version: 1.9.9 (git: 71ebc3e8)

Python: 3.11.5
psutil package: 5.9.5
platform package: 1.0.8
click package: 8.1.7
distro package: 1.8.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.22 %

Total CPU usage: 4.4 %
Total system load: 0.90
Average temp. of all cores: 42.85 °C 

Currently using: schedutil governor
Currently turbo boost is: on

Basically, the CPU temperatures read by auto-cpufreq differ from temperatures read by other software like Psensor, and I think the correct ones are the letters because the the fan are loud.
[image]

shadeyg56, 2023-09-06:
Hi, I have the same issue on my system because `psutil` doesn't return a sensor labeled `CPU`.

Could you do me a favor and enter the following commands into your terminal and posting the results?
```
pip install psutil
python
import psutil
psutil.sensors_temperatures()
```

JosephBoom02, 2023-09-06:
> Hi, I have the same issue on my system because `psutil` doesn't return a sensor labeled `CPU`.
> 
> Could you do me a favor and enter the following commands into your terminal and posting the results?
> 
> ```
> pip install psutil
> python
> import psutil
> psutil.sensors_temperatures()
> ```
`psutil`  in my case gave me the right CPU temperature
`{'nvme': [shwtemp(label='Composite', current=43.85, high=82.85, critical=83.85), shwtemp(label='Sensor 1', current=36.85, high=65261.85, critical=65261.85), shwtemp(label='Sensor 2', current=38.85, high=65261.85, critical=65261.85)], 'k10temp': [shwtemp(label='Tctl', current=60.25, high=None, critical=None)], 'legion_hwmon': [shwtemp(label='CPU Temperature', current=61.0, high=None, critical=None), shwtemp(label='GPU Temperature', current=46.0, high=None, critical=None), shwtemp(label='IC Temperature', current=45.0, high=None, critical=None)], 'iwlwifi_1': [shwtemp(label='', current=44.0, high=None, critical=None)]}
`

shadeyg56, 2023-09-06:
Great, I will work on a fix

PR #548 (2023-09-07): Fix #545: additional cpu temp sensors
This resolves #545.

Not sure how I didn't notice this before, as I was experiencing the same on my laptop.

I believe AMD drivers on Linux do not report individual core temps, so the `coretemp` sensor was not found and it was defaulting to the first sensor it could find which was my SSD temperature.
I added 2 additional sensors to check for. My temp is reported by ACPI, and then k10temp reports the Tctl which is used as a last resort. Now, if it can't find a sensor it will return "nan".
closed 2023-09-07 by AdnanHodzic