AdnanHodzic/auto-cpufreq #552: Overheating of laptop (AMD) while charging

paniash, 2023-09-11:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? Yes
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)? Yes
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)? Yes
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)? Yes

### System information:

Add/paste output of:

```
auto-cpufreq --debug
Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: Arch Linux rolling n/a
Linux kernel: 6.5.2-arch1-1
Processor: AMD Ryzen 7 7730U with Radeon Graphics
Cores: 16
Architecture: x86_64
Driver: amd-pstate-epp

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 4546 MHz
CPU min frequency: 400 MHz

Core    Usage   Temperature     Frequency
CPU0      2.0%        54 °C      1397 MHz
CPU1      0.0%        54 °C       400 MHz
CPU2      1.0%        54 °C       400 MHz
CPU3      1.0%        54 °C       400 MHz
CPU4      1.0%        54 °C       400 MHz
CPU5      0.0%        54 °C       400 MHz
CPU6      1.0%        54 °C       400 MHz
CPU7      0.0%        54 °C      1586 MHz
CPU8      2.0%        54 °C      2793 MHz
CPU9      1.0%        54 °C       400 MHz
CPU10      1.0%        54 °C       400 MHz
CPU11      0.0%        54 °C       400 MHz
CPU12      0.0%        54 °C      2933 MHz
CPU13      0.0%        54 °C       400 MHz
CPU14      1.0%        54 °C       400 MHz
CPU15      0.0%        54 °C       400 MHz

CPU fan speed: 2200 RPM

auto-cpufreq version: 1.9.9 (git: ee888d47)

Python: 3.11.5
psutil package: 5.9.5
platform package: 1.0.8
click package: 8.1.7
distro package: 1.8.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.18 %

Total CPU usage: 0.6 %
Total system load: 0.43
Average temp. of all cores: 54.00 °C

Currently using: powersave governor
Warning: CPU turbo is not available
Currently turbo boost is: off

-------------------------------------------------------------------------------

```

Also please be descriptive about the issue you're reporting, i.e: what you tried & what's the expected behaviour.

---
My Thinkpad E14 laptop with an AMD Ryzen 7730U CPU works well with battery with temps staying around 40 degrees most of the time. However, while charging, the laptop heats up to 55+ degrees on almost slight to no load. I run a minimal environment (dwm + terminal applications + firefox). This is the default behaviour without any config file.

As I tried fiddling around the a custom config file to mitigate this issue, I changed governor to "ondemand" and "powersave" under `[charger]` but even then there is no improvement in the heating issue while charging.

Having gone through a similar issue (#279), even when turbo is off, the heating issue persists. Albeit in that issue the CPU was Intel (not sure if that makes a difference).

My user-defined config file:
```conf
# settings for when connected to a power source
[charger]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor.
governor = powersave

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_min_freq = 800000

# maximum cpu frequency (in kHz)
# example: for 1GHz = 1000 MHz = 1000000 kHz -> scaling_max_freq = 1000000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_max_freq = 1000000

# turbo boost setting. possible values: always, auto, never
turbo = auto

# settings for when using battery power
[battery]
# see available governors by running: cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
# preferred governor
governor = powersave

# minimum cpu frequency (in kHz)
# example: for 800 MHz = 800000 kHz --> scaling_min_freq = 800000
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# to use this feature, uncomment the following line and set the value accordingly
# scaling_min_freq = 800000

# maximum cpu frequency (in kHz)
# see conversion info: https://www.rapidtables.com/convert/frequency/mhz-to-hz.html
# example: for 1GHz = 1000 MHz = 1000000 kHz -> scaling_max_freq = 1000000
# to use this feature, uncomment the following line and set the value accordingly
# scaling_max_freq = 1000000

# turbo boost setting. possible values: always, auto, never
turbo = auto
```

AdnanHodzic, 2023-09-16:
What was the reason for closing and then re-opening the issue?

paniash, 2023-09-16:
Well before posting, I realized I had not changed the AMD p-state CPU driver to acpi-cpufreq. Unfortunately, even after changing to it, I'm facing the same issue.

AdnanHodzic, 2025-03-22:
Can same fix which was used in #464 be used here?