AdnanHodzic/auto-cpufreq #573: Battery charge limmits interfering with discharge detection

ElSargo, 2023-10-04:
My laptop uses charge limit's to prolong the batteries longevity, as a result that status of the battery may be discharging (since no charge is being added), but it is still using AC power.

The current discharge detection does not account for this behavior: [core.py:311](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto_cpufreq/core.py)
```python
if val == "Discharging":
  # we found a discharging battery
  return False
```

This could be fixed by checking the battery current
```python
if val == "Discharging":
	# we found a discharging battery
	try: 
	    with open(Path(power_supply_path + supply + "/current_now")) as f:
	        current = str(f.read()[:-1])
	        # A battery's status might be "Discharging" but that doesn't necessarily mean it's draining
	        if current != "0":
	            return False
	except FileNotFoundError:
	    return False
```
I'm not sure if this is the optimal approach with is why I haven't opened a pr.
[diff.TXT](https://github.com/AdnanHodzic/auto-cpufreq/files/12807572/diff.TXT)

ElSargo, 2023-10-04:
Possibly related to #83

shadeyg56, 2023-10-05:
I have a charge limit set on my laptop so that it doesn't charge past 80% but I haven't had this issue. 
What does running `cat /sys/class/power_supply/BAT1/status` output for you while the laptop is at the charge limit?

Also may I ask what your using to limit the charge? Is it a kernel firmware or something else?

ElSargo, 2023-10-05:
The output is "Not charging", I might be mistaken as to the cause of the issue. I thought that was it becuase `auto-cpufreq --status` shows
```
Battery is: discharging

Setting to use: "powersave" governor
```
The limit is being set by laptop firmware

shadeyg56, 2023-10-05:
Hm yeah not sure why auto-cpufreq is picking up that its discharging

What do you get from `cat /sys/class/power_supply/ACAD/online`?

ElSargo, 2023-10-05:
cat /sys/class/power_supply/ACAD/online gives no such file error
but /sys/class/power_supply/ADP1/online is 1
After taking the latest from source and installing it, it seems to be working now...
Sorry for the inconvinience

closed 2023-10-05 by ElSargo