AdnanHodzic/auto-cpufreq #577: Can't install `auto-cpufreq` on Void Linux

ElijaRock, 2023-10-10:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted?
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

### Error output:
```text
/usr/local/bin/auto-cpufreq: 42: /opt/auto-cpufreq/venv/bin/activate: OSTYPE: parameter not set
```
---

### System information:

Add/paste output of:

```
$ auto-cpufreq --debug
/usr/local/bin/auto-cpufreq: 42: /opt/auto-cpufreq/venv/bin/activate: OSTYPE: parameter not set
```
Tried to use install script as sudo and then run `auto-cpufreq`, but I keep getting this error despite trying to remove and reinstall.

---

ElijaRock, 2023-10-10:
Sorry, I should include the install details:

```
────────────────────────────────────────────────────────────────────────────────────────────────────

Detected Void Linux distribution

────────────────────────────────────────────────────────────────────────────────────────────────────

Setting up Python environment

[*] Updating repository `https://mirrors.servercentral.com/voidlinux/current/x86_64-repodata' ...
[*] Updating repository `https://repo-default.voidlinux.org/current/multilib/nonfree/x86_64-repodata' ...
[*] Updating repository `https://repo-default.voidlinux.org/current/multilib/x86_64-repodata' ...
[*] Updating repository `https://repo-default.voidlinux.org/current/nonfree/x86_64-repodata' ...
Package `python3' already installed.
Package `python3-pip' already installed.
Package `python3-devel' already installed.
Package `python3-setuptools' already installed.
Package `base-devel' already installed.
Package `dmidecode' already installed.
Package `cairo-devel' already installed.
Package `gobject-introspection' already installed.
Package `gcc' already installed.
Package `gtk+3' already installed.

Installing necessary Python packages

DEPRECATION: Loading egg at /opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq-2.0.0+64b11ab0-py3.12.egg is deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use pip for package installation..
Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (23.2.1)
Requirement already satisfied: wheel in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (0.41.2)
DEPRECATION: Loading egg at /opt/auto-cpufreq/venv/lib/python3.12/site-packages/auto_cpufreq-2.0.0+64b11ab0-py3.12.egg is deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use pip for package installation..
Requirement already satisfied: setuptools in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from -r requirements.txt (line 1)) (68.2.2)
Requirement already satisfied: psutil in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from -r requirements.txt (line 2)) (5.9.5)
Requirement already satisfied: click in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from -r requirements.txt (line 3)) (8.1.7)
Requirement already satisfied: distro in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from -r requirements.txt (line 4)) (1.8.0)
Requirement already satisfied: requests in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from -r requirements.txt (line 5)) (2.31.0)
Requirement already satisfied: PyGObject in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from -r requirements.txt (line 6)) (3.46.0)
Requirement already satisfied: charset-normalizer<4,>=2 in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from requests->-r requirements.txt (line 5)) (3.3.0)
Requirement already satisfied: idna<4,>=2.5 in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from requests->-r requirements.txt (line 5)) (3.4)
Requirement already satisfied: urllib3<3,>=1.21.1 in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from requests->-r requirements.txt (line 5)) (2.0.6)
Requirement already satisfied: certifi>=2017.4.17 in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from requests->-r requirements.txt (line 5)) (2023.7.22)
Requirement already satisfied: pycairo>=1.16.0 in /opt/auto-cpufreq/venv/lib/python3.12/site-packages (from PyGObject->-r requirements.txt (line 6)) (1.25.0)

────────────────────────────────────────────────────────────────────────────────────────────────────

installing auto-cpufreq tool

/opt/auto-cpufreq/venv/lib/python3.12/site-packages/setuptools/__init__.py:80: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
!!

        ********************************************************************************
        Requirements should be satisfied by a PEP 517 installer.
        If you are using pip, you can try `pip install --use-pep517`.
        ********************************************************************************

!!
  dist.fetch_build_eggs(dist.setup_requires)
running install
/opt/auto-cpufreq/venv/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

[... 123 lines cut ...]
Best match: idna 3.4
Adding idna 3.4 to easy-install.pth file

Using /opt/auto-cpufreq/venv/lib/python3.12/site-packages
Searching for charset-normalizer==3.3.0
Best match: charset-normalizer 3.3.0
Adding charset-normalizer 3.3.0 to easy-install.pth file
Installing normalizer script to /opt/auto-cpufreq/venv/bin

Using /opt/auto-cpufreq/venv/lib/python3.12/site-packages
Finished processing dependencies for auto-cpufreq==2.0.0+64b11ab0
writing list of installed files to 'files.txt'

────────────────────────────────────────────────────────────────────────────────────────────────────

auto-cpufreq tool successfully installed.

For list of options, run:
auto-cpufreq --help"

────────────────────────────────────────────────────────────────────────────────────────────────────

❯ auto-cpufreq --help
/usr/local/bin/auto-cpufreq: 42: /opt/auto-cpufreq/venv/bin/activate: OSTYPE: parameter not set
```

shadeyg56, 2023-10-11:
Fixed with PR #578, thanks for bringing this up.

You might still be receiving a deprecation warning when running commands but this doesn't affect any of the functionality. It is just a warning. This should be fixed after #576 is merged.

Please pull the latest changes and run the installer script.

PR #578 (2023-10-11): wrapper: switch sh to bash (Void fix)
Fixes #577 

Using sh in the wrapper shebang threw an error, but only on Void for some reason, so it has been exchanged for bash instead. Also fixed another warning that was only showing up on Void
closed 2023-10-11 by shadeyg56