AdnanHodzic/auto-cpufreq #592: /etc/auto-cpufreq.conf working only in --live mode, not if auto-cpufreq runs as service (OpenSuse)

dgrat, 2023-11-01:
If I install the tool with --install, my /etc/auto-cpufreq.conf is basically ignored.
The output says otherwise, but the settings are only applied if I test with --live

```
localhost:~> sudo auto-cpufreq --debug

Using settings defined in /etc/auto-cpufreq.conf file

-------------------------------------------------------------------------------

Linux distro: openSUSE Tumbleweed UNKNOWN version
Linux kernel: 6.4.12-1-default
Processor: 12th Gen Intel(R) Core(TM) i5-1240P
Cores: 16
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 2101 MHz
CPU min frequency: 400 MHz

Core    Usage   Temperature     Frequency
CPU0      1.0%        47 °C      1538 MHz
CPU1      3.0%        47 °C      2101 MHz
CPU2      0.0%        43 °C      1683 MHz
CPU3      2.0%        43 °C      2101 MHz
CPU4      3.0%        42 °C      2101 MHz
CPU5      1.0%        42 °C      2101 MHz
CPU6      1.0%        44 °C      2101 MHz
CPU7      0.0%        44 °C      1658 MHz
CPU8      1.0%        42 °C      2101 MHz
CPU9      0.0%        42 °C      2101 MHz
CPU10      0.0%        44 °C      2101 MHz
CPU11      0.0%        44 °C      2101 MHz
CPU12      0.0%        43 °C      2101 MHz
CPU13      0.0%        42 °C      2101 MHz
CPU14      0.0%        42 °C      2101 MHz
CPU15      1.0%        42 °C      2101 MHz

auto-cpufreq version: (Snap) 2.0.0

Python: 3.10.12
psutil package: 5.9.5
platform package: 1.0.8
click package: 8.1.7
distro package: 1.8.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.04 %

Total CPU usage: 0.6 %
Total system load: 0.95
Average temp. of all cores: 43.31 °C 

Currently using: performance governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

```

dgrat, 2023-11-01:
`# settings for when connected to a power source
[charger]
governor = schedutil
scaling_min_freq = 400000
scaling_max_freq = 2000000
turbo = auto

# settings for when using battery power
[battery]
governor = powersave
scaling_min_freq = 400000
scaling_max_freq = 2000000
turbo = never
`

AdnanHodzic, 2023-11-03:
Have you tried restarting after creating the config file? I remember that for changes to be picked up auto-cpufreq process had to be stopped.

dgrat, 2023-11-07:
Hi there
Restarting the service is not helping. It applies the config only in live mode.

Animeshz, 2023-12-02:
I recently noticed this, after update to `nixos-23.11` channel, (prior `nixos-23.05` was working fine). I'll bisect and see if I can find anything.

dgrat, 2023-12-07:
I tested a little more with current snap image.. When I run 
sudo auto-cpufreq --live
sudo auto-cpufreq --install 
It works upon restart.  After reboot, the settings are not applied anymore. 

It will work again with
sudo auto-cpufreq --remove
sudo auto-cpufreq --live

I think it is pretty reproducible at least on my system

AdnanHodzic, 2023-12-10:
Please make sure to use the [latest version and preferably use auto-cpufreq-installer.](https://github.com/AdnanHodzic/auto-cpufreq/?tab=readme-ov-file#auto-cpufreq-installer)

dgrat, 2023-12-12:
I tested 2.1 and installed from git repo (not snap). Seems to work again ..

AdnanHodzic, 2023-12-12:
Great, closing the issue then.

closed 2023-12-12 by AdnanHodzic