AdnanHodzic/auto-cpufreq #593: Asus ROG - AMD dGPU is up all the time (Hybrid mode) when auto-cpufreq is running

jlrpalomar, 2023-11-03:
This issue was already discussed on https://github.com/AdnanHodzic/auto-cpufreq/issues/278 but closed due to inactivity. Not sure if it's exactly the same root cause, but the consequence is the same.

All the functionality of auto-cpufreq is ok, and it works as intended. However, and I understand that is due to the fact that it has to be monitoring temperature all the time, the dGPU of the computer (when on Hybrid mode) is up all the time.

Running sensors (of lm-sensors package) also does a similar thing, when you run it, to obtain the dGPU temperature, it has to wake it up. I understand that the continuous monitoring of temperature of auto-cpufreq is doing the same, it's getting the temperature of all the sensors of the system, and therefore, requiring the dGPU one, having to wake it up.

I haven't seen any option applicable to configuring auto-cpufreq related with the temperature retrieval. Not very experienced with the psutil package and the methods used by it in core.py but I can give it a try. However I'd appreciate if anyone has more info about the topic.

If I can find anything testing psutil package I'll let it know here.

---

### System information:
-------------------------------------------------------------------------------

Linux distro: Linux Mint 21.2 Victoria
Linux kernel: 6.2.0-36-generic
Processor: AMD Ryzen 9 5900HX with Radeon Graphics
Cores: 16
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3300 MHz
CPU min frequency: 1200 MHz

Core	Usage	Temperature	Frequency
CPU0     12.0%        49 °C      1200 MHz
CPU1      6.1%        49 °C      1200 MHz
CPU2      1.0%        49 °C      1198 MHz
CPU3      0.0%        49 °C      1200 MHz
CPU4      8.2%        49 °C      1200 MHz
CPU5      0.0%        49 °C      1198 MHz
CPU6      2.0%        49 °C      1197 MHz
CPU7     14.7%        49 °C      1197 MHz
CPU8      1.0%        49 °C      1198 MHz
CPU9      0.0%        49 °C      1200 MHz
CPU10      3.0%        49 °C      1200 MHz
CPU11      0.0%        49 °C      1200 MHz
CPU12      7.4%        49 °C      1200 MHz
CPU13      1.0%        49 °C      1200 MHz
CPU14      6.9%        49 °C      1196 MHz
CPU15      0.0%        49 °C      1200 MHz

CPU fan speed: 0 RPM

CPU fan speed: 0 RPM

auto-cpufreq version: 2.0.0 (git: 4ddff1f)

Python: 3.10.12
psutil package: 5.9.6
platform package: 1.0.8
click package: 8.1.7
distro package: 1.8.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.08 %

Total CPU usage: 1.7 %
Total system load: 1.98
Average temp. of all cores: 49.00 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

sayantandas, 2023-11-07:
I can confirm I have the same issue as described above. The system loses battery in 1.5 hours. 
If I use Gnome Power profiles instead I get atleast 3 hours of backup.

```myrog14 ~]$ lspci  | grep -i ATI
01:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Upstream Port of PCI Express Switch (rev c2)
02:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 23 [Radeon RX 6650 XT / 6700S / 6800S] (rev c2)
03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21/23 HDMI/DP Audio Controller
06:00.0 Non-Volatile memory controller: Micron Technology Inc 2450 NVMe SSD [HendrixV] (DRAM-less) (rev 01)
07:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt [Radeon 680M] (rev c8)
07:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt Radeon High Definition Audio Controller```

shadeyg56, 2023-11-08:
I don't have a dGPU so I can't test this for myself, but your explanation makes sense. Unfortunately, `psutil.sensors_temperatures()` only has a parameter for using fahrenheit.

If psutil is indeed causing the problem, it should be possible to ditch psutil in its entirety as all the function does is check `/sys/class/hwmon*` and return the results in a nice format. So we could just create out own temperature function that checks `/sys/class/hwmon*` and only returns the temp for the CPU rather than returning all sensors and grabbing our correct sensor from that list.

AdnanHodzic, 2025-03-22:
Can same fix which was used in #464 be used here?