AdnanHodzic/auto-cpufreq #595: Cannot install auto-cpufreq because of dubious ownership

kchousos, 2023-11-08:
### Fill out information requested in this template, without doing so issue will be ignored & closed!

### Have you tried?

- [Searching through existing/closed issues](https://github.com/AdnanHodzic/auto-cpufreq/issues) to see if your bug has already been already submitted? 
- If installation failed with [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer),have you tried installing auto-cpufreq using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?
- Have you tried [configuring auto-cpufreq](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq)?
- Have you tried suggestions in [troubleshooting section](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting)?

I have tried all relevant things.

### Error output:
```text

Welcome to auto-cpufreq tool installer.
  			
Options:

[I]nstall
[R]emove
[Q]uit

Select a key: [i/r/q]: i

─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

Detected RedHat based distribution

─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

Setting up Python environment

FirefoxPWA                                                                                                                                                                                                   185  B/s | 833  B     00:04    
Package python3-devel-3.12.0-1.fc39.x86_64 is already installed.
Package dmidecode-1:3.5-1.fc39.x86_64 is already installed.
Package gcc-13.2.1-4.fc39.x86_64 is already installed.
Package cairo-devel-1.18.0-1.fc39.x86_64 is already installed.
Package gobject-introspection-devel-1.78.1-1.fc39.x86_64 is already installed.
Package cairo-gobject-devel-1.18.0-1.fc39.x86_64 is already installed.
Package gtk3-devel-3.24.38-3.fc39.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!

Installing necessary Python packages

Requirement already satisfied: pip in /opt/auto-cpufreq/venv/lib64/python3.12/site-packages (23.3.1)
Requirement already satisfied: wheel in /opt/auto-cpufreq/venv/lib64/python3.12/site-packages (0.41.3)

─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

installing auto-cpufreq tool

Processing /home/kchou/Bin/Repos/auto-cpufreq
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [35 lines of output]
      Traceback (most recent call last):
        File "/opt/auto-cpufreq/venv/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/opt/auto-cpufreq/venv/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/opt/auto-cpufreq/venv/lib64/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 149, in prepare_metadata_for_build_wheel
          return hook(metadata_directory, config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/poetry/core/masonry/api.py", line 42, in prepare_metadata_for_build_wheel
          poetry = Factory().create_poetry(Path(".").resolve(), with_groups=False)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/poetry_dynamic_versioning/patch.py", line 26, in alt_poetry_create
          name = _get_and_apply_version(
                 ^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/poetry_dynamic_versioning/__init__.py", line 485, in _get_and_apply_version
          version = _get_version(config, name)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/poetry_dynamic_versioning/__init__.py", line 312, in _get_version
          version = _get_version_from_dunamai(vcs, pattern, config)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/poetry_dynamic_versioning/__init__.py", line 264, in _get_version_from_dunamai
          return Version.from_vcs(
                 ^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/dunamai/__init__.py", line 1934, in from_vcs
          return cls._do_vcs_callback(
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/dunamai/__init__.py", line 1973, in _do_vcs_callback
          return callback(**kwargs)
                 ^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/dunamai/__init__.py", line 1066, in from_git
          _detect_vcs(vcs, path)
        File "/tmp/pip-build-env-zoafuqqy/overlay/lib/python3.12/site-packages/dunamai/__init__.py", line 354, in _detect_vcs
          raise RuntimeError(dubious_ownership_error)
      RuntimeError: Detected Git repository, but failed because of dubious ownership
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

auto-cpufreq tool successfully installed.

For list of options, run:
auto-cpufreq --help"

─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

```
---

### System information:

Running fedora 39.

---

Is this a known issue? Is a fix available?

kchousos, 2023-11-08:
For now I reverted back to commit c3e2f83 before pyproject.toml was introduced.

shadeyg56, 2023-11-08:
Unable to reproduce in a Fedora 39 container. This seems like a local issue related to permissions, but I do find it weird since auto-cpufreq does not make use of any git submodules.

I found [this stackoverflow post](https://stackoverflow.com/questions/72978485/git-submodule-update-failed-with-fatal-detected-dubious-ownership-in-repositor) that points at it being a permission error with the repo directory. Make sure you are signed into the user who owns the directory or take ownership of the directory with `chown`

closed 2023-11-08 by shadeyg56

kchousos, 2023-11-08:
Even after chowing the directory recursively, the problem still persists.

shadeyg56, 2023-11-09:
Hm, from what I can find it's an issue with Git. Check out this link and try this: https://sam.hooke.me/note/2023/08/poetry-fixing-dubious-ownership-error/

AdnanHodzic, 2023-11-12:
Just merged #597 which seems to fix this issue.

kchousos, 2023-11-12:
> Just merged #597 which seems to fix this issue.

This works perfectly! Thanks!